OpenEVSE/openevse_esp32_firmware #122: Base topic changes

robao23, 2020-10-16:
Hi guys.

I have a ghost in open evse I want to know if it is a reported error and if so, what solution does it have.

I put a base topic in services, base topic.

And after a few hours, a few days or a week I just change it.

Example:

246f2893d3dcOpenEVSE

Today I realized that the eco mode was loading at 16A and in open evse it was set this: 

246f2893d3dc2246f2893d3dcOpenEVSE

I am a newbie in this, these are the data that I see on the web, in case they are useful or if they need more information.

WiFi Firmware
ESP8266
Version: 3.1.0.dev

Greetings.

glynhudson, 2020-10-20:
Sorry to hear your having issues. Was the ESP32 module pre-loaded with firmware and purchased via the OpenEnergyMonitor store?

Try updating to the latest version via the web interface: https://github.com/OpenEVSE/ESP32_WiFi_V3.x/releases/tag/3.3.2

robao23, 2020-10-24:
> Sorry to hear your having issues. Was the ESP32 module pre-loaded with firmware and purchased via the OpenEnergyMonitor store?
> 
> Try updating to the latest version via the web interface: https://github.com/OpenEVSE/ESP32_WiFi_V3.x/releases/tag/3.3.2
thanks for answering

Yes, buy the advance kit in June at the official store.

I will try with that new firmware.

It is valid for my system, now it says this;

WiFi Firmware
ESP8266
Version: 3.1.0.

glynhudson, 2020-10-27:
Yes, you have an ES32 module so you can flash the latest V3 firmware. Current latest is V3.3.2 

https://github.com/openevse/ESP32_WiFi_V3.x/releases/

Update can be done via the web interface.

closed 2021-05-06 by glynhudson