OpenEVSE/openevse_esp32_firmware #124: Daylight Saving Time (DST)

glynhudson, 2020-10-21:
DST needs to be implemented for regions that unfortunately needs to deal with DST. Could this be done automatically based on NTP country selection? Or maybe it's easier to implement with a simple check box for 'enable DST'?

jeremypoulter, 2020-10-21:
It should be as far as I am aware... Although - it may not updated the EVSE, may just be the ESP...

glynhudson, 2020-10-21:
Ah interesting, was DST implemented as part of NTP? I assumed the NTP will return unix time

jeremypoulter, 2020-10-21:
Yeah, DST should have been built into the time zone selection, as in if you select Europe/London it should be also applying the DST hour so GMT+1

jeremypoulter, 2020-10-27:
This looked to have worked ok for me, did you see any issues?

glynhudson, 2020-10-27:
Yes, I can confirm DST is already implemented! It worked perfectly for me at the weekend. Nice work 👍

closed 2020-10-27 by glynhudson