OpenEVSE/openevse_esp32_firmware #130: Tesla charging problem in cold weather

Zuikkis, 2020-11-21:
This is not really an OpenEVSE problem, but more likely a problem in recent Tesla firmware. However the way OpenEVSE reacts is a bit worrying and perhaps should be improved.

Tesla seems to have an issue in cold weather. If battery is cold, max charge amps are limited. However when charging starts, it seems to briefly allow too much current, and causes perhaps some protection to kick in which cuts charging completely.

This looks horrible in Domoticz log (OpenEVSE state through MQTT):
[image]

It is banging both the OpenEVSE and Tesla contactors at 5 sec interval. It went on like this for hours before I noticed. Then I checked my logs and noticed that it has happened already a few times earlier, always when it has been below 0C outside!

Now, if I manually set max charge current to 6A, it does start charging normally. Then after a few minutes I can change it to 16A and it will not cut, and Tesla will limit the current if needed (only takes 11-13A or so at 0C).

So.. perhaps there should be some restart delay in case charger is toggling like this? Or just enter some fail mode completely instead of trying again.

Also perhaps a configurable startup ramp, from 6A to max in few minutes or so?

Could be that Tesla fixes this also in some future update. But now I'm very concerned in using the timer as I fear I will wear out the contactors..

jeremypoulter, 2020-11-21:
I am afraid I do not think there is anything we can do from the charger POV. All the charger does is tell the car how much current it can pull, it is down to the car to decide the rate at which it charges.

Zuikkis, 2020-11-21:
Yeah I know how it works. :) OpenEVSE could first advertise 6A, then later ramp to 16A.

chris1howell, 2020-11-21:
It is possible to create a work around to override the vehicle, however
OpenEVSE is currently following the protocol and commands from the vehicle.

The vehicle or responsible to commands the state to start and stop charging
by changing the pilot from 6v to 9v and decides how much current to draw.
OpenEVSE is just responding to the vehicle commands.

chris1howell, 2020-11-21:
If you do choose a work around, I would suggest using MQTT and Node-Red to
create the logic, this way the controller firmware is not modified and will
work as intended.

Once OpenEVSE is connected to a MQTT server, you can create temperature
based logic to start at a low current and increase after a specified amount
of time.

Zuikkis, 2020-11-21:
@chris1howell Thanks that's a good idea, didn't thought of that.

It should be relatively easy. I'm not using OpenEVSE timer, my home automation (Domoticz) controls the charging. It's easy to add a delayed trigger which executes few minutes after charge has started.

I'm closing this issue, I'm happy with the solution. :)

closed 2020-11-21 by Zuikkis