OpenEVSE/openevse_esp32_firmware #133: Simple RFID Authentication

jshum1, 2020-12-08:
I would like to add capability to perform simple access control using RFID cards - I have several neighbors that will use my charger, and would like access control, plus a fairly easy method of querying energy consumption per user.  I have an OpenEVSE kit on the way to be installed next week, and I ordered a simple MFRC522 board to pair with it.  I could probably get it working with a separate Arduino board and MQTT, but it would be nice to interface directly to the ESP32 in the OpenEVSE kit.  If I get a chance to try it out, is there a preferred method of implementing?  Where in the charging flow diagram should access control be inserted?  Also, the board connects via SPI - any issues with that?  There is supposedly a way to force I2C mode out of the board if that's preferable.

jeremypoulter, 2021-02-18:
There is now a PR open to add support for RFID, #146

jeremypoulter, 2021-07-24:
Now #167

closed 2021-07-26 by glynhudson