OpenEVSE/openevse_esp32_firmware #134: Session Log

chris1howell, 2020-12-19:
Log and save each charge session to WiFi Flash.
Plug in - log Start date and time
Un-plug - log Stop date and time time, elapsed time and energy 

Start, Stop, elapsed, energy

jeremypoulter, 2021-02-17:
Can certainly look at this, may need to consider how much history is maintained. Should be ok from the point of view of not wearing out the flash, but maybe a future revision of the WiFI board should include an SD card slot?

chris1howell, 2021-02-18:
Agreed, It will depend on how much space is available for the logs and how
much space is used for each entry. We would not want to erase and rewrite
too often.  Home stations would likely be less than one entry a day.

I am sure there is a better structured and more efficient way to do this...
But Something like this is 71 bytes
{"Start":202101121914,"Stop":202101130215,"Elapsed":601,"Energy":24342}

If we set aside 512KB of the 4MB Flash that would be a little more than
7200 Sessions or 20 years. That is probably a bit excessive to keep on
flash...

chris1howell, 2021-07-19:
OCPP Charge Point operators require Fault and Event logs. 

Event, Session and Faults should be logged to Flash for display in the OpenEVSE Web portal, MQTT and OCPP.

A entry in the log should be created any time the EVSE State changes or Temperature throttling starts/stops

Type - Time - EVSE State - elapsed time - Pilot Current - energy - temperature - temperature max for the current session - Connector/Vehicle  status - Divert mode

Type
1 Information - Session info
2 Notification - Temperature throttling etc.
3 Warning - Errors GFCI, Ground Fault, Temperature Shutdown etc.

Type
1 Information - Session info
2 Notification - Temperture throttoling etc.
3 Warning - Errors GFCI, Ground Fault, Temperature Shutdown etc.

jeremypoulter, 2021-07-25:
In terms of space there is actually not a lot of free flash, OTA requires the same flash space as the app, currently the flash map is:

```text
EEPROM: 20K
OTA data: 8K
app 0: 1.185M 
app 1: 1.185M 
File system: 192K
```

So basically it is a section out of the 192K of the file system we have for storage, still can probably to a significant amount

matth-x, 2021-07-26:
Okay, thank you for the hint, I definetly have to consider this.

From the point of view of the OCPP library, there are internal events (like a successful transaction end) and external events (like throttling due to high temperature). And the same goes for the EVSE state.

I think a human readable plain-text log is more suitable than a well-defined structured log. The format would look like this:
`<type> <timestamp> <internal state> <external state> (<internal event> or <external event>)`, e.g.
`2 | 2021-07-26T13:00:00Z | txId=123, charging | plugged, active, 110°F | end temperature throttling`

This will lead to two additional functions in the API, `log_event(enum type, const char *event)` and `log_state_cb([]() -> const char*)`. And for the European Union, I will introduce preferences to filter out specific internal data if the client has objections regarding the GDPR, which will be set at initiation of the library.

To limit the size of the logs, the plain-text approach requires to chunk the log and delete the oldest chunk if some threshold is exceeded (like occupied memory). When uploading the logs to the server, the chunks will be concatenated to a single file.

Do you agree with that approach or have any improvements in mind?

jeremypoulter, 2021-07-27:
@matth-x I have already started on this from a generic POV. I still think it makes sense to store as JSON lines, is just easier to extract the data back out that way, don't have to write a parser. We don't have to impose a strict format on this. Each record will have some common fields, but nothing to say they all have to have the same data (but probably will).

glynhudson, 2021-10-08:
I've just been testing the event log changes. The event log seems to work really well.

I  think the event log should be hidden away under 'advanced', what do you think @chris1howell? Also the Hardware section should probably be moved to the system tab.

A minor issue is the event log doesn't fit on a mobile device screen very well. The fit could be improved by shortening the text e.g 'information' to 'info'

[image]

glynhudson, 2021-10-19:
I've been testing this for a week or so now. To begin with it seemed to work perfectly, however over the last couple of days it's been incorrectly displaying 0kWh as the session energy. 

The multiple plug / unplug events are due to me using a single EVSE to charge multiple EVs each evening and somtimes a topup during the day. 

<h2 class="left" style="padding-bottom: 0.2em; border-bottom: 1px solid rgb(238, 238, 238); margin-top: 2px; float: left; clear: left; color: rgb(51, 51, 51); font-family: Helvetica, sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">History</h2>

Time | Type | Status | Energy | Temp
-- | -- | -- | -- | --
18/10/2021, 21:25:23 | information | Charging | 0.00 kWh | 39.5 °C
18/10/2021, 21:25:11 | information | EV Connected | 0.00 kWh | 39.5 °C
18/10/2021, 21:12:18 | information | EV Not connected | 4.70 kWh | 40.2 °C
18/10/2021, 20:28:40 | information | Charging | 0.00 kWh | 30.2 °C
18/10/2021, 20:28:38 | information | EV Connected | 0.00 kWh | 30.2 °C
18/10/2021, 20:00:00 | information | EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 16:47:34 | information | Waiting - EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 09:45:43 | information | Waiting - EV Not connected | 0.00 kWh | 30.7 °C
18/10/2021, 07:40:04 | information | Waiting - EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 08:40:01 | information | EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 07:37:59 | information | Waiting - EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 08:37:57 | information | EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 07:00:08 | information | Waiting - EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 08:00:04 | information | EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 06:55:32 | information | Waiting - EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 07:55:30 | information | EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 06:37:31 | information | Waiting - EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 07:37:28 | information | EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 06:32:40 | information | Waiting - EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 07:32:38 | information | EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 05:46:54 | information | Waiting - EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 06:46:51 | information | EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 05:23:39 | information | Waiting - EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 06:23:38 | information | EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 05:12:25 | information | Waiting - EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 06:12:20 | information | EV Not connected | 0.00 kWh | 31.5 °C
18/10/2021, 01:31:51 | information | Waiting - EV Not connected | 0.00 kWh | 48.5 °C
18/10/2021, 01:31:45 | information | Waiting - EV Not connected | 0.00 kWh | 48.5 °C
18/10/2021, 01:30:33 | information | Waiting - EV Connected | 0.00 kWh | 48.5 °C
18/10/2021, 01:30:23 | information | Waiting - EV Not connected | 0.00 kWh | 48.5 °C
18/10/2021, 01:30:01 | information | Waiting - EV Connected | 9.30 kWh | 48.5 °C
18/10/2021, 00:13:42 | information | Charging | 0.00 kWh | 42.2 °C
18/10/2021, 00:13:34 | information | EV Connected | 0.00 kWh | 42.2 °C
18/10/2021, 00:06:33 | information | EV Not connected | 6.75 kWh | 43.0 °C
17/10/2021, 23:03:55 | information | Charging | 0.00 kWh | 29.5 °C
17/10/2021, 23:03:52 | information | EV Connected | 0.00 kWh | 29.5 °C
17/10/2021, 15:59:17 | information | EV Not connected | 13.15 kWh | 46.7 °C
17/10/2021, 15:59:09 | information | EV Connected | 13.15 kWh | 46.7 °C
17/10/2021, 14:32:52 | information | Charging | 3.00 kWh | 36.0 °C
17/10/2021, 14:13:34 | information | Charging | 0.76 kWh | 30.7 °C
17/10/2021, 14:07:08 | information | Charging | 0.00 kWh | 29.5 °C
17/10/2021, 14:07:06 | information | EV Connected | 0.00 kWh | 29.5 °C
17/10/2021, 12:04:47 | information | EV Not connected | 8.00 kWh | 29.2 °C
16/10/2021, 23:36:06 | information | EV Not connected | 8.00 kWh | 44.7 °C
16/10/2021, 23:23:38 | information | Waiting - EV Connected | 8.00 kWh | 44.7 °C
16/10/2021, 23:23:31 | information | Waiting - EV Connected | 8.00 kWh | 44.5 °C
16/10/2021, 23:03:31 | information | Charging | 5.85 kWh | 42.0 °C
16/10/2021, 23:03:27 | information | Charging | 5.84 kWh | 42.0 °C
16/10/2021, 23:02:58 | information | Charging | 5.79 kWh | 42.0 °C
16/10/2021, 22:09:11 | information | Charging | 0.00 kWh | 29.2 °C
16/10/2021, 22:09:09 | information | EV Connected | 0.00 kWh | 29.2 °C
17/10/2021, 01:27:53 | information | EV Not connected | 8.04 kWh | 29.0 °C
16/10/2021, 11:13:07 | information | EV Not connected | 8.04 kWh | 46.7 °C
16/10/2021, 10:30:09 | information | Charging | 7.30 kWh | 46.7 °C
16/10/2021, 02:19:26 | information | EV Connected | 7.30 kWh | 46.7 °C
16/10/2021, 00:04:09 | information | Charging | 0.00 kWh | 46.5 °C
16/10/2021, 00:04:07 | information | EV Connected | 0.00 kWh | 46.5 °C
16/10/2021, 00:00:21 | information | EV Not connected | 13.93 kWh | 47.0 °C
15/10/2021, 21:51:19 | information | Charging | 0.00 kWh | 43.2 °C
15/10/2021, 21:51:17 | information | EV Connected | 0.00 kWh | 43.2 °C
15/10/2021, 21:47:41 | information | EV Not connected | 8.48 kWh | 43.5 °C

closed 2022-05-25 by glynhudson