OpenEVSE/openevse_esp32_firmware #135: Can't reconnect to MQTT after connection timeout

dantam74, 2020-12-30:
I am using the latest version 3.3.2 and have the following issues:

If the connection MQTT times out, it's impossible to get the ESP32 to reconnect using the WebUI. A reboot of the ESP will solve the problem until it times out again.
However a reboot puts the ESP32 in wifi setup-mode, meaning I have to switch to the openevse-local wifi and then reconnect with the wifi.

So in short. I try to use MQTT but after a while the connection times out (it's alive during charging session, but times out when there's no activity). To reconnect to MQTT I need to reboot the Wifi-module but that clears the wifi-connection so I have to set it up again. 

Are there problems with the specific version? I have just started using the MQTT feature so I haven't tried it on other versions.

dantam74, 2020-12-31:
This seems to be an intermittent problem. After the last reboot the ESP32 reconnects automatically to MQTT after a timeout disconnection.
Hold this issue and I'll investigate more to see how and if the problem can be replicated again.
About the wifi disconnect. I suspect that it might be due a weak wifi-signal which puts the ESP32 in AP-mode after a reboot.

dantam74, 2020-12-31:
The problem is re-occuring and I suspect something is not correct or correctly detected in the mqtt.cpp:mqtt_loop() since it should reconnect when disconnected.
This screenshot is what I get in the webui (MQTT settings as normal, but it won't re-connect).
[image]

The mosquitto.log show that reconnects work at first, but after a while it just stops.

```
2020-12-31T10:00:39: New connection from 192.168.25.98 on port 1883.
2020-12-31T10:00:39: New client connected from 192.168.25.98 as openevse-2a68 (p2, c0, k60).
2020-12-31T12:10:19: Client openevse-2a68 has exceeded timeout, disconnecting.
2020-12-31T12:11:12: New connection from 192.168.25.98 on port 1883.
2020-12-31T12:11:12: New client connected from 192.168.25.98 as openevse-2a68 (p2, c0, k60).
2020-12-31T12:49:39: Client openevse-2a68 has exceeded timeout, disconnecting.
2020-12-31T12:49:39: New connection from 192.168.25.98 on port 1883.
2020-12-31T12:49:39: New client connected from 192.168.25.98 as openevse-2a68 (p2, c0, k60).
2020-12-31T13:50:23: Client openevse-2a68 has exceeded timeout, disconnecting.
2020-12-31T13:50:23: New connection from 192.168.25.98 on port 1883.
2020-12-31T13:50:23: New client connected from 192.168.25.98 as openevse-2a68 (p2, c0, k60).
2020-12-31T14:00:28: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
2020-12-31T14:08:10: Client openevse-2a68 has exceeded timeout, disconnecting.
2020-12-31T14:08:10: New connection from 192.168.25.98 on port 1883.
2020-12-31T14:08:10: New client connected from 192.168.25.98 as openevse-2a68 (p2, c0, k60).
2020-12-31T14:11:40: Client openevse-2a68 has exceeded timeout, disconnecting.
2020-12-31T14:11:53: New connection from 192.168.25.98 on port 1883.
2020-12-31T14:11:53: New client connected from 192.168.25.98 as openevse-2a68 (p2, c0, k60).
2020-12-31T14:30:29: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
2020-12-31T14:36:10: Client openevse-2a68 has exceeded timeout, disconnecting.
2020-12-31T14:36:49: New connection from 192.168.25.98 on port 1883.
2020-12-31T14:36:49: New client connected from 192.168.25.98 as openevse-2a68 (p2, c0, k60).
2020-12-31T14:38:23: Client openevse-2a68 has exceeded timeout, disconnecting.
2020-12-31T14:38:23: New connection from 192.168.25.98 on port 1883.
2020-12-31T14:38:23: New client connected from 192.168.25.98 as openevse-2a68 (p2, c0, k60).
2020-12-31T14:49:40: Client openevse-2a68 has exceeded timeout, disconnecting.
2020-12-31T15:00:30: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
2020-12-31T15:30:31: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
2020-12-31T16:00:32: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
2020-12-31T16:30:33: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
```

ruimarinho, 2021-01-09:
I am having exactly the same issue, to the point I am considering downgrading to v2 because v3 has proven to be so unreliable with my MQTT setup.

dantam74, 2021-01-16:
My OpenEVSE spontanously reconnected to Mosquitto after what I think was a reboot. I've noticed that a reboot causes the EmonCMS-app to report the total accumulated kWh as daily usage. However, after a while (this time almost 2 days) the connection was lost again.

Could this problem described here be a similiar problem:
https://stackoverflow.com/questions/65575975/mqtt-client-exceeds-timeout-and-disconnects

jeremypoulter, 2021-01-16:
Would it be possible to get the debug log? http://OpenEVSE/debug

jeremypoulter, 2021-01-16:
Also what are the MQTT setting on the OpenEVSE? Anything special about the MQTT broker?

slavkobrzeg, 2021-01-16:
Leaving the comment to let you know that I have the same problem. Although it seems to work fine since a few days now.

dantam74, 2021-01-16:
> Also what are the MQTT setting on the OpenEVSE? Anything special about the MQTT broker?

You can see the MQTT-settings in my third post of this thread. The MQTT broker is a standard mosquitto v1.6.9. The server disconnects inactive clients after 10 minutes by default. As you can see in the mosquitto log the reconnects from the client side work at first but then it just seems to give up.

dantam74, 2021-01-16:
> Would it be possible to get the debug log? http://OpenEVSE/debug

If I can catch it when it happens I post it here :-)

ruimarinho, 2021-04-11:
I only see `Free memory` messages under `/debug` but I'm still experiencing this problem @jeremypoulter. Anything I can help to debug?

dantam74, 2021-04-12:
This problem still occurs for me but I've found a somewhat "cure". The only way to get the OpenEVSE to connect when it has lost the connection is to reboot the ESP32 (using the reboot button on the Web UI). It can then "hang on" for days or weeks until it looses it again.

jonjander, 2021-04-27:
**My setup**
•	Unifi (UAP AC) accesspoint 2m away (-60dBm)
I have a separate 2g SSID with legacy mode enabled
•	OpenEVSE Wifi firmware 3.3.2
•	OpenEVSE 5.0.1
•	Sending RAPI over MQTT to control $FD, $FE & $SC

**Issue**
The WIFI unit randomly disconnects from my network and fails to reconnect. I have to manually connect to the hotspot network and reconnect (this never fails and the SSID and password is stored I just have to click connect).

Then it can be online for 2m to 15h, then the story repeats.

Today the WIFI disconnected 4 times in a row 4min in between, now its stable.

I have not been able to get any useful debug logs.

ruimarinho, 2021-09-03:
I'm still having exactly the issue @jonjander described, also on Ubiquiti hardware.

rvoosterhout, 2021-09-17:
Yes it looks like this is the same issue. The only difference is i'm not using Unifi AP's. But I don't suspect that those cause the issue. If I can be of any help, or you need remote access to my openvse, let me know.

closed 2023-03-31 by glynhudson

jonjander, 2023-04-21:
I solved it by restarting every day using an external automation, I'm calling http://192.168.1.xxx/restart. 
Ugly workaround :/ so it would be nice if it just reconnected on its own.

KipK, 2023-04-21:
@jonjander, It should be fixed if you upgrade to latest f version.

danepowell, 2023-10-05:
Can you be more specific, i.e. what commit / version fixes this? I'm still experiencing the bug in 4.2.2. Happens whenever my MQTT server restarts.

jeremypoulter, 2023-10-05:
There were a number of network and MQTT fixes in the 5.0.0 release. Give that a try

dmelliott20, 2024-03-27:
Hi I’m experiencing this too - on UniFi kit too

OpenEVSE	8.2.3.EU	
OpenEVSE Wifi v5.1.0_modified

debug logs show nothing around mqtt