OpenEVSE/openevse_esp32_firmware #136: Unable to connect to (non-secure/1883) MQTT broker without enabling "Reject self-signed certificates"

ethan4, 2021-01-05:
I've found that if I try to connect to my broker on the non secure port, with the address set to mqtt NOT mqtts this will fail if I have the check box for "Reject self-signed certificates" disabled.

In the mosquitto log I get a SOCKET error warning. This is also the same warning I get if I try to connect on the MQTTS port which works properly with a valid certificate from Lets Encrypt so perhaps the first time I tried to configure MQTTS and failed there is now a stuck parameter or similar which is causing a problem.

ethan4, 2021-01-05:
See issue #137 for a related problem.

jeremypoulter, 2021-01-05:
Hum... I thought we had fixed this issue... jeremypoulter/ArduinoMongoose#13. Need to check we are using that fix

closed 2023-03-31 by glynhudson