OpenEVSE/openevse_esp32_firmware #140: DEV V4.0: LCD does not display set current in 'Ready' mode

glynhudson, 2021-02-03:
In 'Ready or mode the LCD displays "Ready 32A". However, if the maximum current is adjusted on the interface e.g 6A the LCD still shows 32A. I think this is because the Pilot value still seems to be 32A: 

[image]

TheBluProject, 2021-03-07:
Seems like similiar issue to the lifetime kWh update, it actually gets sent to EVSE but does not show on the GUI until the restart of the WiFi module - seems like the value is only retrieved upon the init / start

glynhudson, 2021-05-06:
This is still an issue in the latest version. It's quite a important and potentially serious issue since being able to see the max current of the unit before connecting can be critical for some use cases and confusing for many others. The LCD max current should update to reflect the max current setting.  

[image]

rbray89, 2021-11-25:
Ah. Grateful to finally see that I'm not crazy. I can confirm this is an issue on the current release. This is rather important for me, as the pilot amperage retrieved from MQTT is never updated, so I can't create my intelligent circuit sharing system until this is resolved.

PR #296 (2022-01-24): Updating the firmware to handle configuring the EVSE
Moved all the 'setting' related RAPI commands from the UI to the backend to fix various issues with the EVSE settings changing without the WiFi module knowing. This was mainly to resolve issues around the current not being displayed correctly on the LCD and in the UI.

- Fixes #140 
- Fixes #178

This change also include some updates to the RAPI tab. You can now click a button and call all the `$Gx` commands to get the current state of the EVSE module.
closed 2022-01-24 by glynhudson