OpenEVSE/openevse_esp32_firmware #143: DEV V4.0: LED on OpenEVSE WiFi does not change color.

chris1howell, 2021-02-05:
LEDs On OpenEVSE Wifi v1 board remain WHITE during boot and in all states.

jeremypoulter, 2021-02-28:
This should be working now

glynhudson, 2021-03-03:
Yes, LED's are now working now. Nice work. 

However, the colour for "sleeping - EV connected" is almost identical to the "charging" colour. This means when a timed charge starts or stops the colour change is indistinguishable. This is also an issue with the LCD backlight and interface colour. Do you notice this @chris1howell? Do you have any thoughts? Could we change the colour for "sleeping - EV connected" to contrast a bit more with "charging"?

jeremypoulter, 2021-03-04:
Yes that is correct, the behavior mirrors the LCD and both states use cyan. The the LCD the only other option is white or blue, all the other colours are used by other states. With th LEDs we have a bit more flexibility

closed 2021-03-16 by chris1howell