OpenEVSE/openevse_esp32_firmware #145: DEV V4.0: Add IP address and host name to LCD cycle

glynhudson, 2021-02-09:
I think it would be useful to display the IP address and the hostname of the EVSE on the menu cycle, what do you think @chris1howell? I know the IP address and hostname are displayed at statup but if the IP address changes or the user forgets the hostname it can sometimes be tricky to power cycle and get back to the display in time to see the current IP address!

chris1howell, 2021-02-17:
I am not sure about this one. Presenting an IP address all the time seems like a security risk.

jeremypoulter, 2021-02-17:
Could use the long press of the button to show the IP address/host name like pressing the button on the WiFi module?

glynhudson, 2021-02-19:
I don't think it's a big security issue since the perpetrator would
need to be connected to your network to be able to access the charge
point, once they are connected IP addresses are trivial to scan.

glynhudson, 2023-03-31:
I think that adding the hostname to the LCD cycle would be a really useful addition, a lot of users struggle to connect to the unit once it's connected to their LAN. I don't think showing the hostname is a security risk e.g `openevse-xxxx`. This would also make it possible to distinguish between units when several units are installed in the same location.

KipK, 2023-05-08:
Perhaps this should be activated by default, and can be disabled from UI config ?