OpenEVSE/openevse_esp32_firmware #147: DEV V4.0: Pysical push button overwride

glynhudson, 2021-02-16:
The physical push button should be able to override the charging timer to start or stop a charge

jeremypoulter, 2021-02-16:
This should be done in the latest code

glynhudson, 2021-02-24:
I've just compiled and tested the latest version but the push button is still unresponsive. Does a vehicle need to be plugged in and a timer set for the button to do anything? Or should the button toggle between Ready > Sleeping > Ready states as before?

jeremypoulter, 2021-02-24:
What OpenEVSE firmware are you using?

glynhudson, 2021-02-25:
Ah I had flashed the latest version to my home EVSE which is still running 7.0.1. 
I'm now tested on my dev bench running 7.1.2 and the override buttons works 👍 

It seems to work fine to toggle between Sleeping/Ready when no timer is set. However, if a timer is set and the EVSE is sleeping pushing the button once just displays "Manual override" but the EVSE stays sleeping. Pushing the button twice in quick succession starts charging momentarily before the EVSE returns to sleeping.

jeremypoulter, 2021-02-28:
The web interface is still talking directly to the controller vi RAPI, this still needs updating to use the new priority system. I added the "Manual Override" text to let the user know that the automated behaviours (PV divert, timers, etc) are no longer functioning and the user has to hit the button again (or use the web interface), While this is slightly different than the existing behaviour it is needed to get some sanity out of multiple things trying to control the EVSE.

glynhudson, 2021-03-03:
Sure, makes sense. Happy to keep the text it will probably make debugging customer issues easier! 

This is still an issue in the latest version > If a timer is set and the EVSE is sleeping pushing the button starts charging momentarily before the EVSE returns to sleeping with "Sleeping | Manual Override" text on the LCD and subsequent button presses are ignored.

glynhudson, 2021-03-06:
I've noticed an issue with the push button: if the push button is pressed to manually start a charging session the unit then remains in manual override mode even after this charging session has finished and all charging timers are ignored. There is no indication on the web UI that the unit is in manual override mode. I think the manual override mode should:

1. Disable and return the normal mode when a charging session finishes (vehicle unplugged) 
2. It should be possible to disable manual override via the web UI and have an indication on the web UI if manual mode is active 

Or an alternative would be just to get rid of manual override mode and just force the push button to toggle the state whatever mode is set and the EVSE would still honour charging timers e.g If a charging timer is set for 2pm-3pm but EV is plugged in a 1pm and the user presses the push button to start immediately the EV will charge from 1pm-3pm when it's stopped by the timer.

jeremypoulter, 2021-03-06:
Sure, I am in the process of adding the API for the GUI to control the manual override, and also clear when the charge session finishes.

glynhudson, 2021-03-12:
Thanks, please make this a priority. I've had two occasions now when my EV has failed to charge overnight via the charging timer since the manual override button has been used earlier in the day. As soon as charging finishes or there is any other state change manual override should be removed.

jeremypoulter, 2021-03-25:
This should be resolved now

chris1howell, 2021-04-29:
A customer has suggested using the button to switch between Eco/Normal,  this seems like a good idea if eco is setup and there are no timers active.

glynhudson, 2021-05-06:
The push button is now working, however I've found it to be intermittent. Sometimes when a charge is active pushing the button just displays "manual override" on the LCD, but the charge does not stop, pushing the button a 2nd time is required to put the unit to sleep to stop a charge.