OpenEVSE/openevse_esp32_firmware #148: V4.0.0 Beta Total Energy Usage not showing.

TheBluProject, 2021-02-18:
After update, it reset the total usage, after manually adding the figure using the RAPI  $SK it still shows zero on the dash
 and in /status json

[image]

 It displays it correctly when queried with RAPI $GU. 

[image]

The debug also shows a zero total.

[image]

GUI does not show the protocol version (displays just "-") 
[image]
RAPI $GV shows protocol version correctly
[image]

glynhudson, 2021-02-18:
Thanks for reporting, we'll investigate this.

jeremypoulter, 2021-02-18:
The protocol version has been deprecated in the [OpenEVSE firmware](https://github.com/lincomatic/open_evse/commit/81d5169c41dc3ea0e3d3e3235171ad1f713ae135) and is not updated in all cases so it has been removed from the HTTP API. Will remove from the GUI soon too.

TheBluProject, 2021-02-19:
As the release note states that minimum OpenEVSE firmware for v4.0.0BETA is 7.1.2, I have updated OpenEVSE to that version, but no change, still exactly the same behaviour.

jeremypoulter, 2021-02-20:
Sorry I am not able to reproduce the total energy bug. 

Are you able to attach the debug output?

TheBluProject, 2021-02-20:
It seems that it only manifests itself after the EVSE FW update when it wipes the flash (and the accumulated watthours), when they are manually added back thru RAPI the are not being displayed (or more to the point retrieved from the EVSE) until the charging process starts. After that it all shows up correctly in the GUI and in status json.

jeremypoulter, 2021-02-20:
Ah ok, that makes sense as the energy stats are only retrieved when
charging, on the assumption that it wouldn't change if not charging.

TheBluProject, 2021-02-20:
Ha, sorry for wasting your time then, next time I will look at the code first ;)

**It's not a  bug, it's a feature**

Perhaps worth adding a short line to readme stating that this is the expected behaviour, as the accumulated gets wiped after the update. Closing

closed 2021-02-20 by TheBluProject

jeremypoulter, 2021-02-20:
It is an unhanded use-case ;-) We maybe could add an API/UI to set the totals, or maybe even store the total on the ESP32 side at the end of each charging session, already a similar ticket....

TheBluProject, 2021-02-21:
I think the way the code is evolving and with potential (highly anticipated) inclusion of the RFID #146 it would make sense to store the charging log as proposed here #134 with the addition of RFID token number or just a token ID if they will be registered withing the ESP32 and then querying it to produce totals / per-period / per-token that might help with the "neighbourly-plug-sharing" scenarios.

P.S.
I am currently experimenting with a few graphic displays + a joystick ranging from OLEDs that fit exactly into the space currently occupied by 16x2 to a monster Inkplate6 that has dual core ESP32 onboard. I assume that there is no will to move all the code for the display (and to connect the display itself) into ESP32 for compatibility and legacy reasons, althought that could help greatly with write-only displays that require a bit more RAM.  Would be nice to get OpenEVSE to look some day like [THAT](https://www.behance.net/gallery/107192937/PILOT-AC-Charging-Pile)

Sorry for the OT ;-P