OpenEVSE/openevse_esp32_firmware #149: Where is Open EVSE firmware 7.2.1 (for 4.0) ?

treepleks, 2021-02-21:
Dear Glyn, Jeremy,

The release notes of the 4.0 beta say " This release requires a minimum of 7.1.2 of the OpenEVSE firmware". But I couldn't find such a release on the Open EVSE GitHub repo (it looks like the latest one is 5.0.1). Could you update the notes with a link to the corresponding firmware repo?

Also, it's unclear if this can be flashed on a good old OpenEVSE hardware with, I guess, an Huzzah32 wifi module and RTC/LCD. A reassuring sentence would be great :-)

TheBluProject, 2021-02-21:
Hi,

You can find version D7.1.2 [here](https://github.com/lincomatic/open_evse)

I am running it on OpenEVSE v4 hardware with Huzzah32 and colour LCD.

HTH

treepleks, 2021-02-21:
Thanks a lot Blu HTH! 

This will be my 1st OpenEVSE flashing. I just compiled it using the openevse_1-6-23 platformio target and I get an ascii firmware.hex file. I understand this cannot be OTA flashed.

Does it need to be flashed through the ISP1 header?

TheBluProject, 2021-02-21:
Hi,
Yes, it requires using the ISP header and ISP Programmer 

You can find a quick syntax for avrdude [HERE](https://github.com/openenergymonitor/open_evse/#falsh-pre-compiled-using-avrdude)

Bear in mind that it erases the flash first, so if you want to retain the accumulated Wh, jot the number down, and then re-add it after flashing using RAPI command $SK. Bear in mind that it will not show on the display or in GUI after being restored until you connect your car and it starts charging, as per #148 .

HTH = Hope that Helps ;)
TBP

treepleks, 2021-02-21:
It does help :-) I will need the header/programmer then...

TBP ! (Thanks Blu Project :-)

closed 2021-02-21 by treepleks

glynhudson, 2021-02-22:
I would only recommend running 4.0 beta if you want to help with testing, it's beta for a reason. If you want a stable system without any bugs I would recommend V3.3.2. 

V7.2.1 controller firmware is also still in dev, use at your own risk. Although it's working fine in internal testing so far.

treepleks, 2021-02-28:
Thanks, Glyn, for the kind warnings. I understand what beta software is and I won't complain if things go amok.

I understand I can flash both the ATMEL and the ESP32 controllers to earlier versions if needed (I have a 6 pins SPI USB flashing device for the ATMEL). I cannot see how I could endanger my car nor my house (I have proper current limiting devices just before the OpenEVSE device and my car will not take more than what it needs).

I won't do this soon anyway. Maybe the software will go out of beta before I do :-)

Have a nice weekend,