OpenEVSE/openevse_esp32_firmware #151: DEV V4.0 Emoncms & MQTT posting

glynhudson, 2021-02-24:
Currently, in the latest dev version I've noticed some issues with MQTT and Emoncms services 

# MQTT

The normal metrics e.g amp, temp etc are not being posted to MQTT. Onle a few new metrics are being posting e.g

[image]

# Emoncms 

After updating from V3.x to V4.x via the web interface I had to disable then re-enable Emoncms to begin posting. Then it posted two messages before stopping and not posting anymore.

[image]
[image]

# New Vehicle state

It's great the new state "vehicle" is being posted to indicate if a vehicle is connected or not. However, please could you change it to be "1" and "0" instead of "true" or "false" since Emoncms only supports numerical values and all other state values are numerical.

jeremypoulter, 2021-02-24:
No probs, can we separate this in to 3 tickets for each different issue?

glynhudson, 2021-02-24:
Sure, good point

jeremypoulter, 2021-02-28:
This is an issue with all the data not being 'collected' when not charging. I made an optimisation to not collect some stats when not charging (like amps) and the stats are only posted to MQTT/EmonCMS when all are collected. Should be an easy fix.

glynhudson, 2021-03-03:
Great, it's working now. RAPI over MQTT is also working 👍

closed 2021-03-03 by glynhudson