OpenEVSE/openevse_esp32_firmware #152: Dev V4.0 RAPI over MQTT not working?

glynhudson, 2021-02-24:
I've noticed it's no longer possible to send RAPI commands over MQTT, is this an intentional change? 

This is a braking change for several applications e.g DemandShaper, OpenDSR etc. We will need to put an alternative API in place.

jeremypoulter, 2021-02-24:
It is not intentional (yet) at some point I think we will need to depricate the RAPI over MQTT (and HTTP) as v4.0 assumes it is the only thing in control of the EVSE (by design)

glynhudson, 2021-02-24:
Sure, I agree this would a good idea. But before it's deprecated we will need to provide a documented MQTT and HTTP API for the basic functions: status, start/stop and set timer. 

It would be good to fix RAPI over MQTT for the latest beta version so it can be used for the moment.

jeremypoulter, 2021-03-02:
I have just been testing this and I can not replicate. RAPI over MQTT looks to work for me:

```
openevse-a7d4/rapi/in/$GV (null)
openevse-a7d4/rapi/out $OK D7.1.2 5.1.3^67
```

chris1howell, 2021-03-02:
I have not tested RAPI over MQTT lately, but I was able to use RAPI via
HTTP today.

A big step to remove RAPI over MQTT would be to push the JSON of /config at
boot and whenever changed. That would kill the need to pull values
individually.

Off topic but the $SC XX M sets the station Maximum and burns to EEPROM
it.is one time and requires a erace.to remove the max. Pretty cool.

glynhudson, 2021-03-03:
Rapi over MQTT is now working. 

Pushing the info in /config to MQTT is a good idea. 

>Off topic but the $SC XX M sets the station Maximum and burns to EEPROM
it.is one time and requires a erace.to remove the max. Pretty cool.

That is cool. Is this in the latest controller FW? Assuming it's documented in the code?

closed 2021-03-03 by glynhudson