OpenEVSE/openevse_esp32_firmware #153: DEV V4.0 Lifetime energy incorrectly displayes 0kWh

glynhudson, 2021-02-24:
[image]

Correct value should be: 

[image]

jeremypoulter, 2021-02-24:
Is this the same issue as #148 ?  If not can you provide the debug/rapi output?

glynhudson, 2021-02-24:
Ah yes it's a duplicate. Just reading your replies to the #148 you mention that energy states are only received when charging. I think this is the issue, I just updated the WiFi firmware on my home EVSE which has a non-zero lifetime energy value. I've not yet started a charge since updated. Presumably the value will update once I perform a charging session? I'll confirm tonight. 

It will be confusing for users updating the WiFi firmware. Does this also mean the lifetime energy display will always display 0Wh after a power cycle?  Ideally the lifetime kWh value would be retrieved from the controller at startup.

jeremypoulter, 2021-02-24:
I think it should be retrieved after boot, I will double check. I have a habit of leaving my dev setup in charging.

glynhudson, 2021-02-25:
I tried a couple of reboots and the lifetime energy value was not retrieved.

After a charging session was started the lifetime kWh was retrieved correctly. However, if the WiFi module is rebooted the lifetime kWh counter reverts to 0kWh until a charging session is started. 

It's important that the lifetime kWh is retrieved at startup.

glynhudson, 2021-03-03:
I've just tested running latest version and after restarting the WiFi module the lifetime total still displays 0 kWh before any charging session occurs.

TheBluProject, 2021-03-04:
Firmware built from the most recent commits for WiFi / GUI / Libs. Both reseting the ESP32 and powering down and powering up the entire unit results in the correct behaviour - lifetime totals as they are suppose to be.

glynhudson, 2021-03-06:
Apologies, I just realised I was seeing 0kWh since I had just flashed new FW to my EVSE which reset the lifetime energy! This is now fixed

closed 2021-03-06 by glynhudson