OpenEVSE/openevse_esp32_firmware #155: DEV V4.0 Vehicle State formatt

glynhudson, 2021-02-24:
It's great the new state "vehicle" is being posted to indicate if a vehicle is connected or not. However, please could you change it to be "1" and "0" instead of "true" or "false" since Emoncms only supports numerical values and all other openevse state values are numerical

TheBluProject, 2021-02-24:
Hi
Would it make sense to actually create additional EVSE state based on the information whether the vehicle is connected or not instead of having another boolean ? We have a slight duplication now as EVSE states already indicate the vehicle connected, the only issue was that when the EVSE was sleeping it was displaying state 254 despite the car connection state. 

Perhaps it would be just nicer to add another EVSE state (10 for the lack of the better number) that is returned when the vehicle = 1 and EVSE is sleeping and call it Sleeping / Connected in the GUI ?

Just my 2¢

glynhudson, 2021-02-24:
Interesting idea, I'm not adverse to that. What do you think? @chris1howell

jeremypoulter, 2021-02-24:
This is how it is implemented on the underlying OpenEVSE firmware and IMHO I don't thinj we should change it, what if the OpenEVSE firmware adds a state 10? Ok we could nullify this by just using a state >0 or <255 but you could also have the case where that same state is added to the controller.

If anything I think we should go the other way and add more abstractions. There is a lot of duplication in the layers of parsing of that state into things like 'is error', 'is charging', 'is connected', 'is sleeping', etc most of those are true under multiple states. The other issue with adding another state is it is not very good for future proofing,every new state has to be correctly interpreted by every client, where as a specific bool for a specific condition is much more future proof.

TheBluProject, 2021-02-24:
Ok, in this case would it be fine to just add the new "state" (without actually assigning the number to it) to the ESP32 purely for it being displayed on the GUI ? At the moment apart from the vehicle being true (or 1 in the near future) when checking the api there's no indication that vehicle is connected when the EVSE is sleeping.

glynhudson, 2021-02-25:
Thinking more about I think your right @jeremypoulter, there is value in having the vehicle connected state in addition to the EVSE state e.g it's possible for there to be a GFCI error or temperature error with the vehicle connector or not connected. There is value in knowing the status of the vehicle connection in every state. Let's go with keeping it the same as it's currently implemented and just change it to boolean numerical. It's also more human-readable then a random number that must be memorized!

glynhudson, 2021-03-03:
This has now been fixed. Thanks @jeremypoulter 👍

closed 2021-03-03 by glynhudson