OpenEVSE/openevse_esp32_firmware #156: DEV V4.0 Temperature display on web interface

glynhudson, 2021-02-25:
After updating to controller V7.1.2 the temperature is no longer displayed on the web interface 

[image]

I think the issue is that temp1 is no longer available: 

[image]

temp4 has also now appeared, I'm not sure where these temperatures are coming from! Presumably one is the LCD and the other is the WiFi module? Is this documented somewhere @chris1howell? Which temp sensor is the best and most consistent to use on the web interface? 

We can assume that users running V4 WiFi will be running controller V7.1.2 or above.

jeremypoulter, 2021-02-28:
It is working for me :( Can you check you have the latest GUI? This should display the temp assigned to `temp`: 
[image]

This is the same as is displayed on the LCD and what will be used for temp throttling. It is selected from essentially the first temp sensor in the first, but we can change the priority if need be. `temp4` is the temp sensor on the WiFi module.

glynhudson, 2021-03-03:
Updated to latest version, temp1 is shown on the web interface.

closed 2021-03-03 by glynhudson