OpenEVSE/openevse_esp32_firmware #157: Github Actions not running?

glynhudson, 2021-03-02:
I've just tried to find the latest binary to test via github actions. However, the last time jeremypoulter/issue5 branch was built was 18 days ago. Should github actions run each time a commit is made? I can see the latest commit to the jeremypoulter/issue5 branch was 2 days ago. 

[image]

jeremypoulter, 2021-03-02:
Hum... not sure... @chris1howell have you made any changes?

chris1howell, 2021-03-02:
I have not made any changes, was wondering the same thing.

jeremypoulter, 2021-03-03:
Hum... that is really odd... Is working fine on master and also if manually triggered just not on the branch...

jeremypoulter, 2021-03-03:
I think I have 'fixed' this, I removed all of the conditions so it runs the builds on all branches/PRs

closed 2021-03-03 by jeremypoulter