OpenEVSE/openevse_esp32_firmware #158: RFC: How best to maintain v3 and v4 in parallel

jeremypoulter, 2021-03-03:
I guess the first question is do we need to maintain v3? The logic being there will be folks that can't/don't want to update the EVSE firmware so we may want to still support them with limited bug fixes/updates that don't depend or the core updates in v4.

So assuming this assertion is true what do we think would be the most user friendly way to do this?

- New 'fork' for v4
- Rename the repo (OpenEVSE_WiFi ?) with v4 on `master` and a `v3` branch for v3? 
- Rename the repo with v3 on `master` and a `v4` branch for v4? 
- Something else?

@glynhudson / @chris1howell but anyone else feel free to add your 2p/2c.

glynhudson, 2021-03-03:
My feeling is that V3 is currently stable and functional, users who don't want to upgrade EVSE FW can stick with V3 but V3 will not get any future updates. I vote for remaining repo to V4 on mater branch can V3 on an archived branch. Going forward we will just mention V4.

chris1howell, 2021-03-03:
> archived branch. Going forward we will just maintain V4.

I agree. Couldn't have said it better Glyn.

jeremypoulter, 2021-03-03:
Thanks, that was probably my preference as well, shall we give it a couple of days for any other comments? I would like to get #139 merged to make the ticket management a little easier

glynhudson, 2021-03-03:
I'm happy for you to go ahead and do it now. The binaries for V3 will remain on the releases page for anyone wanting to use them. We will just need to make clear what controller FW version is required

jeremypoulter, 2021-03-07:
I have created the v3 branch and merged #139 

@chris1howell  Please can you rename the repository and also set master on the GUI repo back to the default branch as I have updated that repo to follow the same pattern, ie a branch for v2 and v3 with v4 on the master.

closed 2021-03-16 by chris1howell