OpenEVSE/openevse_esp32_firmware #159: DEV V4.0 After updating to latest version web interface won't load without incognito mode

glynhudson, 2021-03-03:
After updating to the latest version the web interface won't load unless I use my browsers incognito mode which implies this is some sort of cache issue. Maybe bumping the js version number is needed to force the browser to fetch the new version? 

[image]

jeremypoulter, 2021-03-04:
Can you share a screenshot with the network tab open in both working and failing cases? I have added caching support in v4 as it makes it a lot faster. Also holding shift+clicking the reload button will fix the issue

glynhudson, 2021-03-06:
I'm not able to replicate this any more, presumably my browser has refreshed the cache.

jeremypoulter, 2021-03-07:
Potentially can do ` location.reload(true);` after upgrade ([ref](https://stackoverflow.com/questions/5721704/window-location-reload-with-clear-cache))

glynhudson, 2021-03-12:
After updating to the latest version I've not experienced this issue anymore. I'll close this for the moment

closed 2021-03-12 by glynhudson