OpenEVSE/openevse_esp32_firmware #160: DEV V4.0 MQTT Voltage topic is not followed.

slavkobrzeg, 2021-03-04:
Hey guys! Wonderful job, thanks!!!
Today I realized that even if I have the correct Voltage topic set - I haven't changed it since I use OpeEVSE - the main page shows me 240V which is the default setting when we do not have any real value to track. 
[image]
[image]

jeremypoulter, 2021-03-04:
Hum, weird I was looking at this only last night, I think it was working, I will double check however

slavkobrzeg, 2021-03-09:
The pilot info is also wrong. Doesn't follow the set value. And the MQTT message doesn't contain all information, but I think there is a new separate bug on that.

glynhudson, 2021-03-12:
I confirm I can also replicate this issue. Well spotted. 'Voltage' is always 240V regardless of the MQTT input

ayasystems, 2021-05-05:
Here the same problem. Voltage is always 240

glynhudson, 2021-05-06:
This should be fixed in V4.0.1 , what FW version are you running? https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/4.0.1

ayasystems, 2021-05-06:
Hi

I'm sorry, you are right. I was wrong because voltage is updated only when is charging....

Yes, I'm in 4.1

Should be closed the issue, thanks!

closed 2021-05-06 by glynhudson