OpenEVSE/openevse_esp32_firmware #161: Current section under status page reports incorrect max current on L2.

joshchernoff, 2021-03-06:
Expected: 
Given I'm on the `/#status` page
When I set the max current under the setup section to 40a
Then I expect the current section to reflect the 40a value set under the setup section. 

Observed: 
Given I'm on the `/#status` page
When I set the max current under the setup section to 40a
Then I see the current section always shows max at default 80a as seen below in the provided screen show. 

[image]

glynhudson, 2021-03-07:
This is showing the max current that your unit is capable of. The hard limit is set in the firmware, the maximum hard current on your unit is set to 80A. The hard limit cannot be changed via the interface. You can set a soft current limit by adjusting the "Max Current". 

e.g the EU firmware is set to maximum 32A https://github.com/openenergymonitor/open_evse/blob/master/firmware/open_evse/open_evse.h#L427

chris1howell, 2021-03-07:
SAE J1772 defines the Absolute minimum current as 6A and the Absolute
maximum of 80A. This is different than the pilot setting which is the
current reported to the Vehicle. The Absolute MIN and MAX can be changed in
the source code.

The latest OpenEVSE firmware (7.2.1) has a new command $SC xx M to change
the absolute Maximum from the SAE J1772 to another value, typically the
maximum limit of your hardware. Note for security this is a one time
setting written to EEPROM. The controller EEPROM will need to be fully
wiped and reprogrammed to remove the Limit.

joshchernoff, 2021-03-07:
Closing as not a bug.

joshchernoff, 2021-03-07:
Thanks for the update. It's a bummer the UI does not better convey that as it's easy to conflate the two at first glance.

closed 2021-03-07 by joshchernoff