OpenEVSE/openevse_esp32_firmware #164: MQTT session info missing from periodic status

blobule, 2021-03-09:
The periodic status of MQTT is sending quite a bit of info ( amp, voltage, pilot, wh, state, ...) (see create_rapi_json in src/input.cpp) but some important info is missing:
- session elapsed time (elapsed)
- session energy  (wattsec)
This information is available by issuing rapi commands $GS and $GU, respectively, but issuing these commands forces the MQTT client to be "actively polling" instead of simply (and efficiently) listening to updates.

I wish I could send a pull request, but I can't test the wifi module myself.

glynhudson, 2021-03-12:
The session energy is sent via `wh` 

If you require more information I recommend phasing the json from http://openevse-xxxx/status 

Sending RAPI commands to the controller is not recommended, this will be deprecated very soon.

blobule, 2021-03-18:
The `wh` info is **not** the session energy. It is the **total energy**. It is defined from watthour_total, in input.cpp, function create_rapi_json.

Please double check what you say to justify dismissing an issue. I put a lot of effort to make sure any issue I raise is accurate and verified. I expect those issues not to be dismissed based on a false understanding of what the code does.

Without elapsed/wattsec, I consider MQTT broken. Solving MQTT problems by suggesting to use HTTP is basically suggesting to abandon MQTT.

So I repeat that the session energy in **not** sent through MQTT, and neither is elapsed time. 
The problem can be fixed easily: move elapsed/watthour_total/wattsec from web_server.cpp to input.cpp.
More specifically:
Remove these three lines from web_server.cpp (near line 589):
```
doc["elapsed"] = elapsed;
doc["wattsec"] = wattsec;
doc["watthour"] = watthour_total;
```
Put these same three lines directly inside the function `create_rapi_json()` in `input.cpp` (near ligne 82).
Optionally remove those externals definitions from input.h, as they are now useless globals:
```
extern long elapsed;  // Elapsed time (only valid if charging)
//OpenEVSE Usage Statistics
extern long wattsec;
extern long watthour_total;
```
I' m sorry I can' t submit pull request, as I don' t have a programmer for the esp32. Maybe someday...

glynhudson, 2021-03-26:
Apologies, I misunderstood. Session energy is indeed missing, this would be useful

jeremypoulter, 2021-11-13:
@glynhudson can you check again with the latest dev version?

glynhudson, 2021-11-13:
You're correct, the missing MQTT metrics are present in the latest dev version. This issue can be closed. Thanks a lot.

closed 2021-11-13 by glynhudson