OpenEVSE/openevse_esp32_firmware #165: MQTT rapi commands reply topic hard to identify

blobule, 2021-03-09:
When executing a rapi command sent via MQTT, the reply is sent to <base topic>/rapi/out, which is hard to parse by MQTT since there is no obvious reference to the original command. It is possible to issue a sync number which is sent back but must be parsed, which is tedious.
Could we change the reply topic from **<base topic>/rapi/out** to **<base topic>/rapi/out/<original command>** ?

For example, publishing the message ":12" to "openevse/rapi/in/$GU" will be answered with the message "$OK 40724987 45613 :12^0F"  sent to topic "openevse/rapi/out" , which is then parsed to extract ":12" to figure out that this is the answer to the $GU command we sent.
If we could subscribe to "openevse/rapi/out/$GU" then no parsing for sync number is needed. If someone still want ot subscribe to all answers, then "openevse/rapi/out/#" would work as the original topic.

slavkobrzeg, 2021-03-09:
Hey, I support the need to change the response topic to something that can be easily identified or matched with the just sent request. 
I'm not sure if the proposed solution will work, because I would expect to have a pair of topics for every command. Separately for 'in' and 'out'. This way we can subscribe both and there is no need for guessing. Something like:
openevse/rapi/in/$GU
openevse/rapi/out/$GU

jeremypoulter, 2021-03-09:
This is a fundamental problem with hammering request/response protocols on to MQTT which is really only designed for eventing. 

If you are hitting this sort of error, I strongly suggest using HTTP, the HTTP endpoint can be discovered using MQTT if discovery is difficult in your setup.

Also RAPI comms are deprecated in v4. I will however be making the full HTTP API available over MQTT. What I will probably do is allow the caller to send a token that will also be in the response, which will be a lot more practical as it will (probably) be JSON docs for the request/response.

closed 2021-03-09 by jeremypoulter

blobule, 2021-03-09:
HTTP is very heavy for small clients (say, running on microcontrolers), and managing token is also a burden that can be avoided altogether in MQTT. I don't know what your HTTP api will look like, but I will post a new issue about MQTT.