OpenEVSE/openevse_esp32_firmware #168: version 4.0 with OpenEVSE 5.0.1

ausil, 2021-03-11:
I have firmware 5.0.1 and the ESP32 based wifi hardware running 3.3.2 looking at the release notes for the 4.0 beta it seems I need to update the firmware, though it is not clear where that should come from as the current firmware repo https://github.com/OpenEVSE/open_evse has not had a commit since the 5.0.1 release. I assume I am missing something.

jeremypoulter, 2021-03-11:
I added links to the [readme](https://github.com/OpenEVSE/ESP32_WiFi_V3.x/blob/master/readme.md) but will also update the release notes

glynhudson, 2021-03-12:
OpenEVSE controller version 7.1.2 can be downloaded from here: https://github.com/openenergymonitor/open_evse/releases

closed 2021-03-12 by glynhudson