OpenEVSE/openevse_esp32_firmware #170: V4.0 Bug: Stuck in sleeping mode

glynhudson, 2021-03-12:
I've just updated to the latest V3 version, and I've got a strange issue where the unit is stuck in Sleeping mode, see screencast  

https://drive.google.com/file/d/14KlcV3a_p4Weh1HiUBe_FY-qFcJ3FxxO/view

debug output 

```
{"state":1,"vehicle":0}
{"state":254,"vehicle":0}
{"state":1,"vehicle":0}
{"state":254,"vehicle":0}
Time1
{"amp":0,"voltage":240,"pilot":6,"wh":0,"temp":206,"temp1":false,"temp2":206,"temp3":false,"temp4":317.5,"state":254,"vehicle":0,"colour":5,"freeram":222380,"divertmode":1,"srssi":-72}
{"state":1,"vehicle":0}
{"state":254,"vehicle":0}
{"state":1,"vehicle":0}
{"state":254,"vehicle":0}
{"state":1,"vehicle":0}
Time1
{"state":254,"vehicle":0}
{"amp":0,"voltage":240,"pilot":6,"wh":0,"temp":206,"temp1":false,"temp2":206,"temp3":false,"temp4":315,"state":254,"vehicle":0,"colour":5,"freeram":222340,"divertmode":1,"srssi":-72}
{"state":1,"vehicle":0}
{"state":254,"vehicle":0}
Time1
{"amp":0,"voltage":240,"pilot":6,"wh":0,"temp":207,"temp1":false,"temp2":207,"temp3":false,"temp4":317.5,"state":254,"vehicle":0,"colour":5,"freeram":222312,"divertmode":1,"srssi":-73}
Time1
{"amp":0,"voltage":240,"pilot":6,"wh":0,"temp":207,"temp1":false,"temp2":207,"temp3":false,"temp4":317.5,"state":254,"vehicle":0,"colour":5,"freeram":222432,"divertmode":1,"srssi":-73}
{"state":1,"vehicle":0}
{"state":254,"vehicle":0}
{"state":1,"vehicle":0}
{"state":254,"vehicle":0}
{"state":1,"vehicle":0}
{"state":254,"vehicle":0}
{"state":1,"vehicle":0}
{"state":254,"vehicle":0}
Time1
{"amp":0,"voltage":240,"pilot":6,"wh":0,"temp":208,"temp1":false,"temp2":208,"temp3":false,"temp4":320,"state":254,"vehicle":0,"colour":5,"freeram":222212,"divertmode":1,"srssi":-73}
{"state":1,"vehicle":0}
{"state":254,"vehicle":0}
{"state":1,"vehicle":0}
{"state":254,"vehicle":0}
Time1
{"amp":0,"voltage":240,"pilot":6,"wh":0,"temp":208,"temp1":false,"temp2":208,"temp3":false,"temp4":322.5,"state":254,"vehicle":0,"colour":5,"freeram":220704,"divertmode":1,"srssi":-73}
```

glynhudson, 2021-03-12:
Ah I think I've spotted the issue, for a split second at startup I can see that the controller has an old timer set on it. I've no idea how this timer came to be set, since I've been testing V4 on this setup for weeks. How can I remove this timer? 

See video: https://photos.app.goo.gl/iTKnFp3kXBy5S4mf6

Is it possible to disable RAPI timers on the controller and add a check at startup to check for and remove any pre-set timers. This will be expecailly important for users upgrading to V4 who already have pre-existing RAPI timers set.

MarkNinja, 2021-03-19:
Is there an alternative API to set the timers now that the RAPI timers are not being used?

I use a slider on my home assistant configuration to set the overnight charge parameters, and it currently uses the MQTT RAPI API.

[image]

jeremypoulter, 2021-03-19:
@glynhudson can you share the WiFi log from bootup? This call https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/src/main.cpp#L164 should import the timer and clear from the OpenEVSE

jeremypoulter, 2021-03-19:
More some reason this comment is not showing in the GitHub UI... However
there is a new API to set timers take a look at
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/test/schedule.http

jophut, 2021-07-14:
I think I had the same issue regarding "stuck in sleeping-mode".
First I tried to directly upgrade from firmware 5.0.1 and WiFi 3.3.1 to firmware 7.1.3 and WiFi 4.0.1
I think I resolved my issue by completely wiping the microcontroller's Flash **AND** EEPROM memory, followed by a fresh Flash and EEPROM image from: https://github.com/OpenEVSE/open_evse/releases/tag/7.1.3   

It took me quite some time to find this issue, probably more people could benefit from resolving this issue.

jeremypoulter, 2021-07-18:
I have found the bug here, the number of commands sent on startup is quite a lot, so was exceeding the command queue length, increasing this length solves the issue...

closed 2021-07-22 by glynhudson