OpenEVSE/openevse_esp32_firmware #171: V4.0 esp32-gateway-e

Pukkahq, 2021-03-17:
When I try to load esp32-gateway-e.bin v4 after flashing it keeps going back to the boot loader. I tried using the partitions and bootloader from V3.3.2.

If i load V3.3.2 with the 3 files it flashes and boots ok.

- esp32-gateway-e.bin
- bootloader.bin
- partitions.bin

Pukkahq, 2021-05-05:
here is the serial output after flashing the latest firmware 

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

glynhudson, 2021-05-06:
Have you seen: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/docs/wired-ethernet.md

This works for me 

`esptool esptool.py --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x8000 partitions.bin 0x10000 esp32-gateway-e.bin`

Pukkahq, 2021-05-08:
If I flash with this config version 3.3.2 works ok, when i change the firmware to version 4.0.1 the unit does not boot

[image]

Pukkahq, 2021-05-09:
The latest release loaded ok.

glynhudson, 2021-05-18:
Good to hear. There is a known issue with HTTP update on the Ethernet gateway https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/187, the workaround is to load manually via micro usb.