OpenEVSE/openevse_esp32_firmware #174: Solar PV divert 'minimum charge time' not taking effect

glynhudson, 2021-03-24:
https://community.openenergymonitor.org/t/openevse-eco-mode-minimum-charge-time/17277/4

pdhoogh, 2022-09-03:
There may be more to this. 

In March '21 @jeremypoulter said:
"_In this condition that is not the min charge time that would apply, the minimum change time starts when the charging starts and ensures that the charge does not stop within the that first 600 seconds (10 mins) as the charge session is going on for longer than 10 mins then the charge will stop when demand goes under the 6A threshold._"

That statement does not align with my observation. I frequently see the following:
[image]

You can see that the charge current is zero, the contactor is not closed. But the countdown on the minimum charge time is running. I am also on 600 seconds minimum charge time. It showing 380 while charge current is zero should be enough evidence to prove my statement: the countdown timer is started when the OpenEVSE thinks it can start the charge - enough current available - **but not that charging actually started**. 

It is not because OpenEVSE thinks it can charge that the EV agrees and wants to take the charge. We can talk about following standards, but that does not change the following reality that still happens to me with the Peugeot e-208:

- OpenEVSE sees enough current available and starts the countdown timer for minimum charge time. Even if the EV is seen as "disconnected".
- The EV does not take the charge but the countdown timer continues
- At some point the EV decides it is going to take the charge. The countdown timer almost ran out.
- The EV initializes charging, the contactor closes
- The countdown timer runs out and solar deflates before the car finished initializing the charge session
-  EVSE opens the contactor because the timer ran out and available current is less than 6 Amps before the car finished initializing the charge. 
- The EV reports a charger error and disables the charge port. 
- The only way to reset this is to physically go and unplug the EV and plug it back in. 

If the minimum charge time is set to 60 seconds, that happens a lot with cloudy skies or washing machines switching the heating on and off. 60 seconds is not enough for the e-208 to fully initialize the charge.

So I'd suggest two improvements:

1. Start the minimum charge countdown timer only when the contactor really closed or, even better, the charge current has reached 6 Amps or more, i.e. confirmation charging is initialized by the EV
2. Set a minimum clamp on the minimum charge time setting or inform the user that the setting must be higher than the time the car needs to initialize the charge. That may not be necessary is in 1. the option is chosen to start the minimum charge timer only when the charge current reached 6 Amps

Hope this helps.

jeremypoulter, 2022-09-03:
That is very interesting and I think you are correct. I will take a look.

PR #431 (2022-10-12): Divert updates
Various updates and improvements to help fix issues around not stopping PV diversion when expected as well as some general improvements and tidy ups

Fixes #387 
Fixes #361
Fixes #362
Fixes #174
Possible fix for #388
closed 2022-10-12 by glynhudson