OpenEVSE/openevse_esp32_firmware #175: Implement OCPP protocol V1.6

glynhudson, 2021-03-24:
Open Charge Point Protocol: http://www.openchargealliance.org/protocols/ocpp/ocpp-16/

Allow OpenEVSE to integrate easily with other charging networks. 

Spec attached: 
[ocpp-1.6_edition_2.pdf](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/files/1408057/ocpp-1.6_edition_2.pdf)
[OCPP_1.6_JSON_Specification.pdf](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/files/1408058/OCPP_1.6_JSON_Specification.pdf)

Wiki page: https://github.com/OpenEVSE/openevse_wifi_server/wiki/Open-Charge-Point-Protocol

EvChargepoints, 2021-03-30:
Hi Glynn I hope all is well. Was you aware we was copied in to this ? 

Best Regards 

Paul Delahay

jeremypoulter, 2021-03-31:
This was just moved from the v2 project and you had commented on that thread. As you are still alive, please can you pay the nearly £2000 you owe me?

chuck-h, 2021-05-07:
@matth-x Props for your work on this!!
I am curious where current control over OCPP (e.g. reduce charging current if electrical service is overloaded) fits in to your plans. I have started experimenting with this on the host side using a SteVe add-on. The OCPP 1.6 Smart Charging support (SetChargingProfile etc) seems awfully complex.
I may be able to help with implementing it if desired.

matth-x, 2021-05-08:
Smart Charging was actually the central concern of this project when I started it. The library fully supports the `SetChargingProfile` operation for the single connector use case. You can find an example [here](https://github.com/matth-x/ArduinoOcpp/blob/be15473ef90fb94ea1c3a6ef9c886946ceba8a5f/examples/ESP32/main.ino#L48) on how to integrate that functionality into your charger.

Btw. thank you for the nice comment. If you have further questions, don't hesitate to ask (also possible via [private message](https://github.com/matth-x/ArduinoOcpp#further-help)). If you like to share your ideas, I would be curious to learn about them.

glynhudson, 2021-07-23:
Thanks to @matth-x for implementing basic OCPP implementation in https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/182

The latest version is available in V4.1.0 release (currently in beta).

Testing feedback of the beta would be appreciated, it's probably best to open new issues for specific issues from now on: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/4.1.0

closed 2021-07-23 by glynhudson