OpenEVSE/openevse_esp32_firmware #176: NodeMCU32-S environment gets stuck in boot loop, fine in others.

DrFrankReade, 2021-04-04:
I was compiling the V3 firmware for a NodeMCU32-s and it seems to get stuck in a boot loop saying **Factory Reset** no matter what hardware I actually compile it on. Bad ESP probably ruled out.  When I compiled for a Feather ESP32, naturally it howled about some WIFI LED over the USB serial console, but it compiled and connected and ran with the OpenEVSE via the RX2 and TX2 serial pins just fine. I have no idea what it's all about, but it's a thing.  I tried it in this newfangled V4 branch and it kept rebooting on a detached ESP32-NodeMCU32-S as well ( I never hooked it to the OpenEVSE as I'm not ready to flash new firmware to the ATmega328)

So there it is.

jeremypoulter, 2021-04-04:
That will be the button config, is detecting the press on the wrong high/low, needs swapping.

DrFrankReade, 2021-04-04:
That'll do it. Is that the sort of thing that can be put on an internal pull-up / pull-down for the disconnected scenarios? It's not like power draw is an issue.

closed 2021-05-06 by glynhudson