OpenEVSE/openevse_esp32_firmware #177: WiFi scanning does not always present list of available networks

ruimarinho, 2021-04-04:
[image]

Using v3.3.2, network scanning does not always present the list of available networks. One can see that the API is returning the correct results, but they are not displayed on the UI.

Sometimes the list is populated with a single entry and then disappears.

jeremypoulter, 2021-04-04:
Can you grab the scan results that produce no entries? I would suspect that they are all have a low signal so are filtered out

ruimarinho, 2021-04-10:
@jeremypoulter it's likely that, yes. Probably missing an empty states that says ("No networks found with signal below 80dBm").

```[{
    "rssi": -77,
    "ssid": "a",
    "bssid": "xxx",
    "channel": 6,
    "secure": 3
}, {
    "rssi": -79,
    "ssid": "b",
    "bssid": "xxx",
    "channel": 6,
    "secure": 3
}, {
    "rssi": -80,
    "ssid": "c",
    "bssid": "xxx",
    "channel": 6,
    "secure": 3
}, {
    "rssi": -82,
    "ssid": "d",
    "bssid": "xxx",
    "channel": 6,
    "secure": 3
}]
```