OpenEVSE/openevse_esp32_firmware #178: Sensor Value: Pilot not reflecting right value > 11A

lebsd, 2021-04-06:
Not certain what is the source data for the value Pilot under the Sensor block display but seems to be inconsistent and perhaps misleading since there is no sensor I am aware for the Pilot.

While operating in Eco Mode, with no timers, and a maximum Value set to 16A under Setup, noticed that the Session current reflects ~16A, but the Sensor Pilot Value seems not to increase above 11A. Checked my Tesla for charging current and it shows 16A.    Also Current Value under Session Status ~16A.

glynhudson, 2021-05-05:
Is this still an issue in the latest 4.01 version? https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/4.0.1

I've not noticed this issue.

jeremypoulter, 2021-05-05:
I think it will be, several bits of the UI are still getting values directly from the EVSE via RAPI, including the pilot value. As this may not be updated as the value is changed by the WIFi module it is not guaranteed to be correct

glynhudson, 2021-05-06:
Ah ok, I am able to replicate the issue. I think we should remove the 'pilot' reading from the sensors display since as the OP mentions it's not really a sensor and is not really needed. Please ignore the reading it for the moment.

glynhudson, 2021-05-06:
Related issue to https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/140

PR #296 (2022-01-24): Updating the firmware to handle configuring the EVSE
Moved all the 'setting' related RAPI commands from the UI to the backend to fix various issues with the EVSE settings changing without the WiFi module knowing. This was mainly to resolve issues around the current not being displayed correctly on the LCD and in the UI.

- Fixes #140 
- Fixes #178

This change also include some updates to the RAPI tab. You can now click a button and call all the `$Gx` commands to get the current state of the EVSE module.
closed 2022-01-24 by glynhudson