OpenEVSE/openevse_esp32_firmware #179: OpenEVSE Firmware EU only?

kallisti5, 2021-04-07:
The new OpenEVSE Firmware referenced [here](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/V4.0.0.BETA) are labeled as *.EU

https://github.com/openenergymonitor/open_evse/releases

* V7.1.3.EU,
* V7.1.2.EU,
* etc.

Are there an US versions? (does it matter?)

I'm assuming the OpenEVSE firmware at https://github.com/openenergymonitor/open_evse/releases are compatible with OpenEVSE Plus v5?   This stuff is getting a bit unclear with the new openenergymonitor repo.

kallisti5, 2021-04-10:
It does appear it matters...

* Disable AUTOSVCLEVEL (autodetection is designed for split-phase)

Could OpenEVSE publish a US build with AUTOSVCLEVEL enabled?  There's no bug tracker there... so this is as close as I can get :-)

kallisti5, 2021-04-11:
Ah.  So my confusion came from the [4.0.0 release](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/V4.0.0.BETA) of openevse wifi references the European OpenEVSE firmware

* The "7.1.2" link goes to https://github.com/openenergymonitor/open_evse/releases
* In master however, the link was corrected to go to: https://github.com/lincomatic/open_evse

Tldr; the "murican" version is https://github.com/lincomatic/open_evse which has an unpublished 7.x version.  The OpenEVSE folks say the official 7.1.x version is targeted in the "next few days"

glynhudson, 2021-04-29:
The EU version will work in the USA and the OpenEnergyMonitor version is compatible with OpenEVSE Plus v5

closed 2021-04-29 by glynhudson