OpenEVSE/openevse_esp32_firmware #180: Support for cheap hardware : WT32-ETH01

robotsflo, 2021-04-10:
Hi,

I would like to compile this code for this cheap board ~7€ .

I have found from tasmota github this :
 
```
//#define USE_ETHERNET                             // Add support for ethernet (Currently fixed for Olimex ESP32-PoE)
//  #define USE_WT32_ETH01                         // Add support for Wireless-Tag WT32-ETH01
//  #define ETH_TYPE          0                    // [EthType] 0 = ETH_PHY_LAN8720, 1 = ETH_PHY_TLK110, 2 = ETH_PHY_IP101
//  #define ETH_ADDRESS       1                    // [EthAddress] 0 = PHY0 .. 31 = PHY31
//  #define ETH_CLKMODE       0                    // [EthClockMode] 0 = ETH_CLOCK_GPIO0_IN, 1 = ETH_CLOCK_GPIO0_OUT, 2 = ETH_CLOCK_GPIO16_OUT, 3 = ETH_CLOCK_GPIO17_OUT
  // Olimex ESP32-PoE
  #define ETH_TYPE          0                    // [EthType] 0 = ETH_PHY_LAN8720, 1 = ETH_PHY_TLK110, 2 = ETH_PHY_IP101
  #define ETH_ADDRESS       0                    // [EthAddress] 0 = PHY0 .. 31 = PHY31
  #define ETH_CLKMODE       3                    // [EthClockMode] 0 = ETH_CLOCK_GPIO0_IN, 1 = ETH_CLOCK_GPIO0_OUT, 2 = ETH_CLOCK_GPIO16_OUT, 3 = ETH_CLOCK_GPIO17_OUT
  // wESP32-PoE
//  #define ETH_TYPE          0                    // [EthType] 0 = ETH_PHY_LAN8720, 1 = ETH_PHY_TLK110, 2 = ETH_PHY_IP101
//  #define ETH_ADDRESS       0                    // [EthAddress] 0 = PHY0 .. 31 = PHY31
//  #define ETH_CLKMODE       0                    // [EthClockMode] 0 = ETH_CLOCK_GPIO0_IN, 1 = ETH_CLOCK_GPIO0_OUT, 2 = ETH_CLOCK_GPIO16_OUT, 3 = ETH_CLOCK_GPIO17_OUT
```

Can you help me ? 

Thanks a lot

jeremypoulter, 2021-04-11:
Probably the closest build we have at the moment is the Olimex ESP32 Gateway: <https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/platformio.ini#L236-L259> you could start by trying to modify that, looks like the main thing you would need to add is `-DETH_ADDRESS=1`. That board also doesn't look like it has a button or LED so the `-DWIFI_....` lines can be removed. 

The other thing that probably needs doing is setting up the serial to use the correct pins, and on that subject the ESP32 is 3.3v and the OpenEVSE module is 5v so technically you need to level shift when connecting the two (the Huzzah and OpenEVSE WiFi boards have level shift built in), connect the two directly at your own risk. That being said I have never had any issues.

ayasystems, 2022-03-02:
Hi

I was working on this device. I think it is working fine....

Could you add an specific env to add WT32-ETH01 @jeremypoulter ??

With this configuration is working

[env:openevse_nodemcu-32s]
platform = ${common.platform}
board = nodemcu-32s
framework = arduino, espidf
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
src_build_flags =
  ${common.version}.dev
  ${common.src_build_flags}
  ${common.debug_flags}
  -D WIFI_LED=12
  -D WIFI_LED_ON_STATE=LOW
  -D RAPI_PORT=Serial2
  -D WIFI_BUTTON=4
build_flags =
  ${common.build_flags}
  -D RX2=5
  -D TX2=17 
  -D"ENABLE_WIRED_ETHERNET"  
  -D"ETH_PHY_POWER=16"
   -D"ETH_PHY_TYPE=ETH_PHY_LAN8720"
   -D"ETH_PHY_ADDR=1"
   -D"ETH_PHY_MDC=23"
   -D"ETH_PHY_MDIO=18"
   -D"ETH_CLK_MODE=ETH_CLOCK_GPIO0_OUT"  
#upload_port = openevse.local
upload_speed = 921600
monitor_speed = 115200
extra_scripts = ${common.extra_scripts}
board_build.partitions = ${common.build_partitions}

[image]

[image]

RAPI PORT is ussing RX TX labeled on board.

Best regards

jeremypoulter, 2022-03-02:
Bit tricky for me to add as I don't have the hardware to test/validate, if possible please create a PR that you have tested

ayasystems, 2022-03-02:
@robotsflo could you test these settings??

robotsflo, 2022-03-02:
Hi, 
Thanks a lot for the recent comment in this issue.
 
I have compile fine the latest version with this settings. 
The upload was succes but i dont have the openevse at my house to test it. 
It's in my parent's house. 

The logs are : 
"OpenEVSE not responding or not connected"

The webui responding but not loading because it.

I can check it again during the next month.

robotsflo, 2022-03-02:
[image]

ayasystems, 2022-03-02:
Check GND RX and TX

[image]

try to change RX/TX...

robotsflo, 2022-03-02:
i will test it soon

closed 2022-05-25 by glynhudson

jdgarcia99, 2023-09-30:
Hi. Did you ever make it work? what about current version as of Sept 2023?
Thanks