OpenEVSE/openevse_esp32_firmware #181: Update the OpenEVSE firmware

jeremypoulter, 2021-04-10:
I need to check the voltage levels but in theory it should be possible to program the firmware of the OpenEVSE from the WiFi module via the larger expansion connector.

It has SPI and a spare gpio for the reset. Also this library, <https://github.com/lbernstone/ESP_AVRISP>, looks like it will do the job.

chris1howell, 2021-04-16:
In theory, a future version of the WiFi board could have the reset pin of the serial port connected to a GPIO. If a bootloader were added to OpenEVSE, firmware could be pushed over serial via the normal reset/serial upload. 

The question is do we want to allow this? The current approach is to require physical connection for safety firmware which rarely changes and allow Wifi/UI changes Over the Air. It may be wise to add additional security precautions before implementing OTA updates to the OpenEVSE controller.

glynhudson, 2021-04-16:
This would be *really* useful, but I agree it could have safety issues. Maybe only default official github release builds of OpenEVSE FW would be allowed via the web upload.

jeremypoulter, 2021-04-17:
I was thinking we could probably build the EVSE firmware into the WiFi
image, like the GUI. Didn't suggest it originally as I figured it wouldn't
of be too inflexible... The other issue is the different firmwares between
US/EU ideally we should make a universal firmware, but probably have the
space to include both.

The other (probably best) option is to digitally sign the firmware, but
that is quite a lot of work to setup.

KipK, 2023-03-24:
shaven't saw this one. Following from https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/577 , seems we don't need a reset pin wired with optiboot nowdays.

Since optiboot 8 you can software reset and jump to bootloader by clearing MCUSR :
https://github.com/Optiboot/optiboot/blob/6f98751c6d184ec4a9c7f1fe0f0dcf12f9db2cc9/optiboot/bootloaders/optiboot/optiboot.c#L628

This needs to be tested but should works for fw update.

We have ton of space left now we can remove timers & limits to put back the bootloader

closed 2026-09-21 by github-actions[bot]