OpenEVSE/openevse_esp32_firmware #184: Solar PV divert race condition?

doppiaemme, 2021-05-03:
Hi, 
I've set up solar PV divert with Grid export, in my scenario I have a Tesla powerwall, and I export the grid +/- via MQTT through homeassistant.
In theory, EV charging should start only when the battery is fully charged, the site consumption is less than the solar production and exporting to the grid.
In reality, what often happens is that, even when the battery is not fully charged, openevse kicks-in and starts charging (haven't understood why!).
But from this moment on, it will never go back. 
The Tesla gateway will never charge the battery again, because Openevse is autoadjusting using all the available power. 
From the Tesla GW perspective infact, it is the site using all the available PV production, so it won't charge the battery.
Not sure what is the best way to fix it, but perhaps adding a field that reads the battery percentage and allows EV charging only when battery is >98% f.i.?
Any suggestion or obvious mistake in my setup?

jeremypoulter, 2021-05-03:
Hum, interesting, as you say probably a race condition between the Power wall and the OpenEVSE.

I think in the first instance you could try making the divert attack ratio lower to make the OpenEVSE slower to respond and start charging. Also does the Grid I/E include the battery charging, ie when charging the battery will the Grid I/E be ~0?

doppiaemme, 2021-05-03:
Yes, the grid value is reported at the edge toward the grid, so yes the battery is included, when charging the export is 0.
But there's something weird to be honest on the whole divert calculations.
For instance, the limit is set to 6A, but it's charging at 32 A:
[image]

So what happens is that not only it stops the battery from being charged, but it is draining it too:

[image]

Can this be a bug?

chris1howell, 2021-05-08:
I use this python code to reserve power to charge the powerwalls if below a certain SOC and publish the values to MQTT. I set solar diver to Solar/PV-GEN and point to the /excessPower topic.

`
import time
import requests
import json
import paho.mqtt.client as mqtt
import urllib3
urllib3.disable_warnings(urllib3.exceptions.InsecureRequestWarning)
 
powerwall_ip = "192.168.1.2"        # local IP address for Tesla energy gateway
powerwall_email = "name@domain.com" # local email for Tesla energy gateway
powerwall_password = "password"     # local password for Tesla energy gateway

mqtt_broker = "mqtt.domain.com"     #MQTT broker
mqtt_port = 8883                    #MQTT port standard 1883 or secure 8883
mqtt_user = "username"              #MQTT username
mqtt_password = "password"          #MQTT password
mqtt_topic = "Tesla_Energy/"        #MQTT base topic

baseline_load = 400                 # watts of baseline always on houshold load
target_soe = 90                     # battery % to target + and minus 5%

def on_connect(client, userdata, flags, rc):     # Setup for MQTT connection
    if rc==0:
        client.connected_flag=True 
    else:
        print("Bad connection Returned code=",rc)
mqtt.Client.connected_flag=False 

payload = {'username': 'customer', 'email': powerwall_email, 'password': powerwall_password, 'force_sm_off': False}
r = requests.post('https://' + powerwall_ip + '/api/login/Basic', verify = False, data = payload) # Authenticate to Tesla Energy Gateway

client = mqtt.Client('openevse') 
client.username_pw_set(username = mqtt_user,password = mqtt_password)    #For MQTT connections with username and password # out if no username/password
client.on_connect=on_connect  
client.connect(mqtt_broker, mqtt_port)
client.loop_start()

print("Connecting to broker ",mqtt_broker)
client.connect(mqtt_broker)     
while not client.connected_flag: 
    time.sleep(1)
while client.connected_flag:
    soe = requests.get('https://' + powerwall_ip + '/api/system_status/soe', verify=False, cookies = r.cookies)       #HTTP GET /api/system_status/soe
    meters = requests.get('https://' + powerwall_ip + '/api/meters/aggregates', verify=False, cookies = r.cookies)    #HTTP GET /api/meters/aggregates

    soe.json()
    soe_json = json.loads(soe.text)                 #parse JSON
    meters.json()
    meters_json = json.loads(meters.text)       # parse JSON
    
    site_json = meters_json["site"]                #Extract site JSON
    battery_json = meters_json["battery"]          #Extract battery JSON
    load_json = meters_json["load"]                #Extract load JSON
    solar_json = meters_json["solar"]		   #Extract solar JSON

    battery_soe = int(soe_json["percentage"])      #Extract Values
    site_power = int(site_json["instant_power"])
    battery_power = int(battery_json["instant_power"])
    battery_voltage = int(battery_json["instant_average_voltage"])
    battery_freq = int(battery_json["frequency"])
    load_power = int(load_json["instant_power"])
    solar_power = int(solar_json["instant_power"]) # End Extract Values

    if battery_soe < target_soe - 5: # Check Battery % reserve power to charge if below target
      print ("Charging Powerwall " + str(battery_soe) + "% " + str(battery_power) + " watts")
      if battery_power > 0:
        excess_power = solar_power - battery_power - 1800
      else:
        excess_power = solar_power - 1800
    if battery_soe > target_soe + 5: # Check Battery % alocate additional power to discharge if abeve target
      print ("Discharging Powerwall" + str(battery_soe) + "% "+ str(battery_power) + " watts")
      if battery_power > 0:
        excess_power = solar_power - battery_power + 200
      else:
        excess_power = solar_power + 200
    if battery_soe >= target_soe - 5 and battery_soe <= target_soe + 5: # Check Battery % reserve exact power to maintain if near target
     print ("Maintaining Powerwall" + str(battery_soe) + "% " + str(battery_power) + " watts")
     if battery_power > 0:
        excess_power = solar_power - battery_power - baseline_load
     else:
        excess_power = solar_power - baseline_load
	 
    print ("Grid Power   " + str(site_power) + " watts") # Print latest values
    print ("Solar Power  " + str(solar_power) + " watts")
    print ("Excess Power " + str(excess_power) + " watts")
	 
    ret= client.publish(mqtt_topic +"soe", battery_soe)  # Publish latest values to MQTT stats
    ret= client.publish(mqtt_topic +"sitePower", site_power)
    ret= client.publish(mqtt_topic +"batteryPower", battery_power)
    ret= client.publish(mqtt_topic +"loadPower", load_power)
    ret= client.publish(mqtt_topic +"solarPower", solar_power)
    ret= client.publish(mqtt_topic +"excessPower", excess_power)
    ret= client.publish(mqtt_topic +"voltage", battery_voltage)
    ret= client.publish(mqtt_topic +"frequency", battery_freq)
    print ("Published to MQTT")
    print ()
    time.sleep(5)
	
client.loop_stop()    #Stop loop 
client.disconnect() # disconnect
`

doppiaemme, 2021-05-26:
@chris1howell wouldn't it be easier to simply add a threshold (read via MQTT), in OpenEVSE, so that it cannot start charging if the battery value is <= min_battery_threshold.
if the value is > than the threshold use the standard divert algorithm

chris1howell, 2021-05-26:
It depends on your priorities. I want a balanced approach. I charge my
powerwall at at the same time a the vehicle so at the end of the day both
the powerwall and the vehicle have as much energy as possible. Hopefully,
this leaves enough energy in the powerwall to last overnight and some added
range for the vehicle.

A minimum would be simple to implement, with the current divert the
powerwalls would stay at the minimum while a vehicle is charging. If the
minimum is set high the powerwall is the priority, the vehicle may never
charge. If the priority is low the vehicle is the priority and the
powerwall may not get a charge.

doppiaemme, 2021-05-27:
yes it depends on personal priority, 
for me the battery is #1, once fully charged, whatever is exported goes into the car.
As openevse for now doesn't take into account the battery I have for now working around that in a similar way than your script, adjusting the MQTT published value:
`- platform: template
  sensors:
    pv_divert_now:
      unit_of_measurement: "W"
      value_template: >- 
        {% if (states('sensor.powerwall_charge') |float) > (states('input_number.battery_threshold') |float) %}
          {{ (((states('sensor.powerwall_site_now')) |float)*1000) | round (0)}}
        {% else %}
            0
        {% endif %}`

closed 2021-07-28 by glynhudson