OpenEVSE/openevse_esp32_firmware #186: V4.0.1 Olimex ESP32 Gateway Not responding OpenEVSE firmware 7.1.3

wjhrdy, 2021-05-07:
V4.0.0 seems to be working but when I upgrade my ESP32 to V4.0.1 it doesn't seem to be connecting to the OpenEVSE Board, luckily I can downgrade back to 4.0.0 and it still works.

OpenEVSE Not responding or not connected

[image]

[image]

glynhudson, 2021-05-08:
Apologies, there was an issue with the 4.0.1 release for the es32 gateway, this has now been fixed. Please to update again using this file: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/download/4.0.1/esp32-gateway-e.bin

Note: currently HTTP update via the web interface is not working for the ESP32 gateway. You will need to update using a micro USB connection and esptool https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/docs/wired-ethernet.md#uploading-pre-compiled

closed 2021-05-08 by glynhudson

wjhrdy, 2021-05-08:
This worked. Thanks.