OpenEVSE/openevse_esp32_firmware #187: HTTP update failes on ESP32 Etherent gateway

glynhudson, 2021-05-08:
Testing trying to update from 3.3.2 to 4.0.1 

[image]
[image]

FatJuan, 2021-05-15:
Just upgraded to 4.0.1 from 3.3.1 using openevse_huzzah32.bin had same issue but resolved itself when tried 2nd time all up to date now. Thanks all!

glynhudson, 2021-05-16:
Interesting, are you using an esp32 Ethernet gateway? Probably not since you mention you are using `openevse_huzzah32.bin`. This issue is referring specially the esp32 Ethernet gateway. HTTP update should be working fine on the Huzzah and OpenEVSE Wi-Fi module.

FatJuan, 2021-05-16:
Sorry misread the header. Yes not using Ethernet gateway.

jeremypoulter, 2021-11-10:
Just been having a look at this, I think the issue is that with the wired connection the update is being delivered to fast and we are never going round the main loop and the watch dog is kicking in and rebooting. Over WiFi the update looks to be working for me.

jeremypoulter, 2021-11-11:
I think this may need some updates to the Mongoose library, we are currently using v6 but there is v7 now. It may be wise to update but this might be a large amount of work...

PR #415 (2022-08-22): Issue #187 - Feed watchdog in http_update_write()
I've only spent a couple minutes testing this on an ESP32-PoE-ISO board, but it seems to work, and it compiles OK for the targets on the releases page.

Fixes #187
closed 2022-08-22 by jeremypoulter