OpenEVSE/openevse_esp32_firmware #189: Add support for (physically) locking the EVSE

jeremypoulter, 2021-05-09:
My Mum's EVSE has a feature where you can lock/unlock the EVSE with a key. This both enable/disables the EVSE and (in theory, doesn't on Mum's) locks the cable in place.

We could support this directly on the OpenEVSE WiFi via the GPIO16 (although we may wish to use that for the reset for #xxxx) or as an alternative to the push button on the OpenEVSE (will need a change to the OpenEVSE firmware).

Locking the Type 2 connector (I assume) could entirely be done in hardware.

jeremypoulter, 2021-05-09:
Example of key switch: https://uk.rs-online.com/mobile/amp/p/key-switches/0562606/

jeremypoulter, 2021-05-09:
Example of locking Type 2 connector: https://ecoharmony.co.uk/products/viridianev-32socket-2-motor

jeremypoulter, 2021-05-09:
(these are not recommendations, just first find on Google as an example)

jeremypoulter, 2021-05-19:
The key lock is to enable/disable charging, adding the mechanical locking is just an added bonus.

glynhudson, 2023-03-31:
RFID functionality which is now available is probably an easier way to achieve the same result.

closed 2023-03-31 by glynhudson