OpenEVSE/openevse_esp32_firmware #191: GUI not available after Wifi restart

doppiaemme, 2021-05-26:
hi, I am running ht latest wifi FW + latest openevse fw.

when I click on "restart" Wifi, I can see it actually does something, but the GUI becomes unavailable afterwards and pinging the unit looses a lot of packets (see below)
To fix it, I need to powercycle the whole openevse HW.

[image]

any clue?
thanks

glynhudson, 2021-05-27:
What hardware and firmware version are you using? 

Can you access the status page e.g http://<openevse-IP>/status

doppiaemme, 2021-05-27:
emonEVSE running 4.0.1 and 7.1.3 (EU).

I can't access anything, the only way to make it work is to power off and on a few times and randomly it becomes responsive.

Is it normal that after a power cycle it doesn't show anymore on the LCD the IP address and host name?
It goes straight into "Ready" mode.

FatJuan, 2021-06-01:
I am having these problems too. I thought it was my wifi at home (has been problematic). That is now sorted. The the GUI becomes unresponsive and when you can get to it it will hang as it loads. (1/17 etc) using 7.1.3.EU and 4.0.1.  It is problematic to change charge rates when you do get through and the solar divert will often not update and will stay with orange figures for 25mins? I realise all this is work in progress and I am grateful for all you do for us!

FatJuan, 2021-06-01:
Also when changes are made (go from solar to manual) get Config failed to save.

doppiaemme, 2021-06-01:
We have very similar issues! The Solar Divert doesn't work at all, it never triggers, even if there is a lot of power available.
Even the switch "normal" "eco" seems not to be working as it should... It's quite useless at the moment.

doppiaemme, 2021-06-01:
> Also when changes are made (go from solar to manual) get Config failed to save.

I had the same but after trying a few restarts it worked again. The issue is that now you don't know what it may happen when you restart, as it may get unresponsive, so I fear doing that!

lmsi57, 2021-06-02:
The same for me
Somebody know if the next release is soon?
[image]

jeremypoulter, 2021-06-02:
When it gets stuck on 0/17 there is normally a JavaScript error, can you check the console in the dev tools

FatJuan, 2021-06-02:
Mine has gone totally unresponsive.  I had to go to unit to override manually.  The date showed correctly, time was 2hr forward.  I did have same sort of problems with an earlier iteration.  I am using the Huzzah ESP32 as supplied in shop.

FatJuan, 2021-06-02:
I do not know if this is any help this is the debug? I have managed to get manual charging.
Time1
{"amp":17600,"voltage":239.22,"pilot":17,"wh":48705,"temp":450,"temp1":450,"temp2":false,"temp3":false,"temp4":false,"state":3,"vehicle":1,"colour":6,"manual_override":1,"freeram":216620,"divertmode":2,"srssi":-61}
{"emoncms_connected":1,"emoncms_message":"null"}

lmsi57, 2021-06-02:
Interesting @FatJuan 
Could you confirm that pilot sensor change when you change value by $SC api command?

FatJuan, 2021-06-02:
I used $SC12 (to set to 12a?) I got >$NK^21. The sensor did not change.

FatJuan, 2021-06-02:
Then it just stopped communicating 
OK 17600 239940^15
$FP 0 0 Charging 17.60A^62
$OK^20
$FP 15 0       ^26
$OK^20
$FS^31
$OK^20
$GS^30
$OK fe 13483 03 0540^1C
$FB 6^36
$OK^20
$FP 0 0 zzZ Sleeping Zzz^3D
$OK^20
$FP 0 1 Time  1:17:30 PM^2F
$OK^20
$FP 0 1 Time  1:17:31 PM^2E
$OK^20
$FP 0 1 Time  1:17:32 PM^2D
$OK^20
$FP 0 1 Time  1:17:33 PM^2C
$OK^20
$FP 0 1 Time  1:17:34 PM^2B
$OK^20
$FP 0 1 Date 02/06/2021^02
$OK^20
$FP 15 1       ^27

lmsi57, 2021-06-02:
> I used $SC12 (to set to 12a?) I got >$NK^21. The sensor did not change.

You need to add a space between C and 1

FatJuan, 2021-06-02:
I did send another command and got reply >$OK 8^38 pilot did set to 8a.  The wifi and the system has a complete mind of its own at the moment.  Works fine then no connection for an hour and then back again.

FatJuan, 2021-06-02:
Enable Solar PV Divert is not updating as it always fails to save config so reduced to trying to manually track sun!

doppiaemme, 2021-06-02:
I wonder if these are wifi firmware issues or the openevse one? 
If it is known what was the latest onw thst worked I would like to roll back as with this latest combination the system is pretty unusable.

FatJuan, 2021-06-03:
Same issue on other thread Can't charge after upgrade to firmware with V5 hardware and Huzzah ESP32 module #193
Must be hardware related?

doppiaemme, 2021-06-03:
I have OpenEVSE V1 HW and similar issues which I didn't have before the upgrades. I don't think it's related to a specific HW model

glynhudson, 2021-06-03:
Are you sure you uploaded the correct FW for the Huzzah? This is `openevse_huzzah32.bin` https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/download/4.0.1/openevse_huzzah32.bin

`firmware.bin` is for the OpenEVSE WiFi V1 module.

[image]

FatJuan, 2021-06-03:
Pretty sure I did.  I have just reflashed, had similar issue to last time firstly "HTTP Update failed"
Then repeated and "Firmware update completed ok"
Right at the moment everything seems OK but I am not charging ATM. Lack of sun.
Grateful for all the help!

FatJuan, 2021-06-03:
So now have sun. Webpage system fails to refresh / load. When it does load it is frozen. When it returns the Solar divert does not update I have bright sun Solar: 841W - 159s | Charging from solar | Charge rate: 3A 
Voltage: 242.6V | Available Current: 3.49A | Smoothed Current: 12.25A I can change charge rate ok via RAPI.

js1tr3, 2021-06-03:
I did flash the wrong firmware at first, but it didn't take. Flashed the correct one on the 2nd go round. Keeping the arduino openevse firmware at 7.1.3, I went back to wifi 3.3.2 and everything worked as expected. Upgraded to 4.0.1 again and no dice.
Do I need to do a factory reset when upgrading the firmware?

FatJuan, 2021-06-03:
Well I could change charge rate via RAPI now Failed to save config.  Refresh page, server stopped responding. After 30 minutes comes back but hangs on Loading, please wait... (0/17) then when loaded Solar divert not updated for 24 minutes but now after 43minutes of no response all updating fine! Now unresponsive again. Now ok I have managed to get the config if of use?
{"firmware":"7.1.3.EU","protocol":"-","espflash":4194304,"version":"4.0.1","diodet":0,"gfcit":0,"groundt":0,"relayt":0,"ventt":0,"tempt":0,"service":2,"scale":220,"offset":0,"mqtt_supported_protocols":["mqtt","mqtts"],"http_supported_protocols":["http","https"],"ssid":"Useless","pass":"_DUMMY_PASSWORD","www_username":"+++++++++++++","www_password":"_DUMMY_PASSWORD","hostname":"openevse-ed60","sntp_hostname":"pool.ntp.org","time_zone":"Europe/London|GMT0BST,M3.5.0/1,M10.5.0","emoncms_server":"http://192.168.1.113/emoncms","emoncms_node":"openevse-ed60","emoncms_apikey":"_DUMMY_PASSWORD","emoncms_fingerprint":"","mqtt_server":"192.168.1.113","mqtt_port":1883,"mqtt_topic":"openevse-ed60","mqtt_user":"emonpi","mqtt_pass":"_DUMMY_PASSWORD","mqtt_solar":"emon/emonpi/elec","mqtt_grid_ie":"","mqtt_vrms":"emon/emonpi/vrms","mqtt_announce_topic":"openevse/announce/ed60","ohm":"","divert_PV_ratio":1.1,"divert_attack_smoothing_factor":0.4,"divert_decay_smoothing_factor":0.05,"divert_min_charge_time":600,"tesla_username":"","tesla_password":"","tesla_vehidx":-1,"led_brightness":128,"flags":1547,"emoncms_enabled":true,"mqtt_enabled":true,"mqtt_reject_unauthorized":true,"ohm_enabled":false,"sntp_enabled":true,"tesla_enabled":false,"divert_enabled":true,"pause_uses_disabled":false,"mqtt_protocol":"mqtt","charge_mode":"eco"}
Any advice?

glynhudson, 2021-06-17:
Yes, it's  a good idea to fully erase the flash and start from fresh if you are having issues  `$ esptool erase_flash`

What hardware module are you using?

js1tr3, 2021-06-17:
Huzzah ESP32, I flashed after erase_flash. solar divert worked flawlessly for a week then quit again today, and I couldn't even charge with eco off.

glynhudson, 2021-06-18:
Could you post a screenshot of the UI to illustrate the issue? 

Did you flash the pre-compiled V4.0.1 huzzah firmware? https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/download/4.0.1/openevse_huzzah32.bin

chris1howell, 2021-06-21:
I just flashed back and forth between several versions and experienced this
issue. I was able to resolve this issue by clearing the browser cache.

treepleks, 2021-06-27:
I have reflashed my ATMEL with 7.1.3 a few weeks ago together with my Huzzah ESP32 (checked identity visually by opening the wallbox, taking a picture with my phone for better accessibility). I flashed from 3.3.2 to 4.0.1. I did use the `openevse_huzzah32.bin` version. Since then, I have inconsistent behavior. The solar divert usually starts fine but has difficulty stopping when the sun goes down. It often continues to charge while the advertised charge rate slowly decreases to 6A, 5A, 4A then ultimately 0A, but the pilot remains at 6A and the relay remains closed. It sometimes works though, but I cannot find why it works or not. The ESP32 also often disconnects from Wifi by itself and shifts to AP mode. It remains in AP mode. I have to make it reconnect manually to the router (although it has the SSI and password: I just have to press the Connect button). The reaction to RAPI commands by MQTT is also inconsistent. Sometimes it works, sometimes the RAPI command is just ignored (possibly, my MQTT divert/voltage updates are too frequent, they come in every 2"? But this worked fine in 3.3.2). Even the manual charging mode may have issues: the screen goes green for 2 seconds and then immediately goes blue again (waiting for I don't know what: no timer, not in divert mode). Finally, last symptoms, there are often configuration saving issues (config failed to save) and this sometimes leads to the wifi module rebooting and remaining stuck in AP mode, again.

I will flash back to 3.3.2 and see how it works. If it works, maybe a (very slow) git bisect could help?

js1tr3, 2021-07-05:
That sounds like the issues I've been having. I'm back on 3.3.2 and that seems to be working for now.

tschiex, 2021-07-05:
You are lucky. Following flashing back to 3.3.2, I still have issues. I'm puzzled. May be it's matter of protocol inconsistencies between the ESP32 and the ATMEL or I have hardware issues. I have recompiled the 4.0.1 firmware on my own and reflashed. Will see.

chris1howell, 2021-07-05:
Try clearing the browser cache. Solved all my issues after flashing several
versions.

tschiex, 2021-07-05:
I'm using various browsers on various computers (Linux, android,...). I will definitely try to do this on all browsers, but wifi de-connections occurs even when I'm not on a browser.Do you think the rev3 UI code (js,...) is still hanging there?

tschiex, 2021-07-05:
I'm still puzzled by the fact that the ESP32 does not seem to be reading the ATMEL protocol correctly under 4.01. In the Hardware panel of the OpenEVSE main tab it says:

```
OpenEVSE   
Firmware:  7.1.3
Protocol:  -
```
Do you all see a dash/minus sign in the protocol? When I use $GV in RAPI, I correctly see `>$OK 7.1.3 5.1.3^22`

treepleks, 2021-07-11:
Clearing the browser cache did not improve things, so I reflashed the ATMEL this morning. Did this from Linux using the following command lines:
```
avrdude -c usbasp -p m328p -P usb -U lfuse:w:0xFF:m -U hfuse:w:0xDF:m -U efuse:w:0x05:m 
avrdude -V  -c usbasp -p m328p -P usb -U flash:w:open_evse.hex 
avrdude -V -c usbasp -p m328p -P usb -U eeprom:w:eeprom_24.bin 
```
avrdude complains on efuse reading 0xFD instead of 0x05. I understand this is expected since only the 3 least bits are valid here. Few hours have passed now and things are looking better. No reboot, and the charging started when needed.

treepleks, 2021-07-17:
Well, few days have passed now and the Open EVSE wall-box seems to be running fine now. I apparently messed up my first flashing session. Don't know what happens. BTW: the protocol version of the ATMEL firmware is still not there (but this is a minor issue :-)

Sorry for the time I took. Keep up the good work!

FatJuan, 2021-07-17:
Still having problems connecting.  Always on the sunniest days when I need it! Clear cach different browsers painful.

js1tr3, 2021-07-17:
I've had luck with incognito mode in firefox.

FatJuan, 2021-07-17:
Just a thought, when I replaced my ESP8266 which needed a drop down unit to get to 5v I think, I just put the newer ESP 32 Huzzah as a replacement.  The newer unit will run from 12v I think but when I received the newer unit it was bare so I just put it in as a direct replacement hence off 5v.  Any chance this is the root of my problems?

glynhudson, 2021-07-22:
I think WiFi fix now merged into master may help with this issue: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/209

We hope to do a new official release soon which will include this change.