OpenEVSE/openevse_esp32_firmware #193: Can charge

js1tr3, 2021-06-02:
Upgraded to 7.1.3 and WiFi 4.0.1
Can’t charge ev bouncing between state 2 and 254.
Rolled back to WiFi 3.3.2 and everything works without downgrading avr firmware.

jeremypoulter, 2021-06-02:
Can you provide the debug and RAPI logs? And maybe the config as well?

js1tr3, 2021-06-02:
I notice the protocol is blank....
[image]

debug 
e":0,"grid_ie":255,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":245,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":243,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":250,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":254,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":245,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":240,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
Time1
{"state":254,"vehicle":1,"colour":6,"manual_override":0}
{"amp":0,"voltage":240,"pilot":8,"wh":0,"temp":285,"temp1":285,"temp2":280,"temp3":false,"temp4":false,"state":254,"vehicle":1,"colour":6,"manual_override":0,"freeram":220124,"divertmode":2,"srssi":-52}
{"divert_update":0,"grid_ie":256,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":260,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":265,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":242,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":235,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":237,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":225,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":228,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":228,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":216,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":218,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
{"divert_update":0,"grid_ie":215,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
Time1
{"amp":0,"voltage":240,"pilot":8,"wh":0,"temp":285,"temp1":285,"temp2":280,"temp3":false,"temp4":false,"state":254,"vehicle":1,"colour":6,"manual_override":0,"freeram":214048,"divertmode":2,"srssi":-51}
{"divert_update":0,"grid_ie":215,"charge_rate":0,"voltage":240,"available_current":0,"smoothed_available_current":0,"pilot":8}
Set charge mode to fast
charge_mode changed
config_divert_enabled() = 1
config_charge_mode() = 0
{"divertmode":1,"divert_active":false}
{"state":2,"vehicle":1,"colour":3,"manual_override":0}
{"divert_update":0,"grid_ie":207}
{"divert_update":0,"grid_ie":202}
{"divert_update":0,"grid_ie":184}
{"divert_update":0,"grid_ie":185}
Time1
{"state":254,"vehicle":1,"colour":6,"manual_override":0}
{"amp":0,"voltage":240,"pilot":8,"wh":0,"temp":285,"temp1":285,"temp2":280,"temp3":false,"temp4":false,"state":254,"vehicle":1,"colour":6,"manual_override":0,"freeram":204432,"divertmode":1,"srssi":-52}
{"divert_update":0,"grid_ie":184}
{"divert_update":0,"grid_ie":165}
{"divert_update":0,"grid_ie":146}
{"state":2,"vehicle":1,"colour":3,"manual_override":1}
{"divert_update":0,"grid_ie":141}
{"divert_update":0,"grid_ie":135}
Time1
{"state":254,"vehicle":1,"colour":6,"manual_override":1}
{"amp":0,"voltage":240,"pilot":8,"wh":0,"temp":285,"temp1":285,"temp2":280,"temp3":false,"temp4":false,"state":254,"vehicle":1,"colour":6,"manual_override":1,"freeram":204728,"divertmode":1,"srssi":-52}
{"divert_update":0,"grid_ie":124}
{"divert_update":0,"grid_ie":108}
{"divert_update":0,"grid_ie":109}
{"divert_update":0,"grid_ie":99}
{"divert_update":0,"grid_ie":81}
Time1
{"amp":0,"voltage":240,"pilot":8,"wh":0,"temp":285,"temp1":285,"temp2":280,"temp3":false,"temp4":false,"state":254,"vehicle":1,"colour":6,"manual_override":1,"freeram":209484,"divertmode":1,"srssi":-51}
{"divert_update":0,"grid_ie":81}
{"divert_update":0,"grid_ie":69}
{"divert_update":0,"grid_ie":82}
{"divert_update":0,"grid_ie":74}
{"divert_update":0,"grid_ie":65}
Time1
{"amp":0,"voltage":240,"pilot":8,"wh":0,"temp":285,"temp1":285,"temp2":280,"temp3":false,"temp4":false,"state":254,"vehicle":1,"colour":6,"manual_override":1,"freeram":209488,"divertmode":1,"srssi":-52}
{"divert_update":0,"grid_ie":63}
{"divert_update":0,"grid_ie":84}
{"divert_update":0,"grid_ie":79}
{"divert_update":0,"grid_ie":65}
{"divert_update":0,"grid_ie":179}
[... 210 lines cut ...]
$OK^20
$FP 15 1       ^27
$OK^20
$SV 245800^0A
$OK^20
$FP 0 1 Manual Override^3D
$OK^20
$FP 15 1       ^27
$OK^20
$SV 245877^0A
$OK^20
$GS^30
$OK fe 0 02 0701^17
$GP^33
$OK 285 280 -2560^29
$FB 6^36
$OK^20
$FP 0 0 zzZ Sleeping Zzz^3D
$OK^20
$FP 0 1 Manual Override^3D
$OK^20
$FP 15 1       ^27
$OK^20
$SV 246118^09
$OK^20

js1tr3, 2021-06-02:
one more log

OpenEVSE WiFI 2b8c
Firmware: 4.0.1
Build date: Apr 29 2021 21:50:59
IDF version: 3.30300.190916
Free: 301180
ssid changed
pass changed
time_zone changed
mqtt_server changed
mqtt_topic changed
mqtt_pass changed
mqtt_grid_ie changed
divert_PV_ratio changed
divert_attack_smoothing_factor changed
divert_decay_smoothing_factor changed
flags changed
After config_load_settings: 298576
Connecting to SSID: intothesea
Got Network event SYSTEM_EVENT_WIFI_READY
Got Network event SYSTEM_EVENT_AP_STOP
Got Network event SYSTEM_EVENT_STA_START
Set host name to openevse-2b8c
Got Network event SYSTEM_EVENT_STA_START
Set host name to openevse-2b8c
After net_setup: 240332
Server started
After web_server_setup: 234760
{"state":254,"vehicle":1,"colour":6,"manual_override":0}
first read RAPI values
Checking for existing timers
start_hour = 0
start_minute = 0
end_hour = 0
end_minute = 0
No timers on EVSE board
{"amp":0,"voltage":240,"pilot":8,"wh":0,"temp":285,"temp1":285,"temp2":280,"temp3":false,"temp4":false,"state":254,"vehicle":1,"colour":6,"manual_override":0,"freeram":229812,"divertmode":1,"srssi":0}
Got Network event SYSTEM_EVENT_STA_CONNECTED
Connected to intothesea
Got Network event SYSTEM_EVENT_STA_GOT_IP
Connected, IP: 192.168.1.38
{"time":"2021-06-02T14:57:50Z","offset":"-0400"}
{"divert_update":0,"grid_ie":118}
{"divert_update":0,"grid_ie":97}
{"divert_update":0,"grid_ie":97}
{"divert_update":0,"grid_ie":103}
Time1
{"amp":0,"voltage":240,"pilot":8,"wh":0,"temp":285,"temp1":285,"temp2":280,"temp3":false,"temp4":false,"state":254,"vehicle":1,"colour":6,"manual_override":0,"freeram":213856,"divertmode":1,"srssi":-52}
{"divert_update":0,"grid_ie":84}
{"state":2,"vehicle":1,"colour":3,"manual_override":1}
{"divert_update":0,"grid_ie":79}
{"divert_update":0,"grid_ie":75}
{"divert_update":0,"grid_ie":63}
{"divert_update":0,"grid_ie":60}
{"divert_update":0,"grid_ie":61}
Time1
{"state":254,"vehicle":1,"colour":6,"manual_override":0}
{"amp":0,"voltage":240,"pilot":8,"wh":0,"temp":285,"temp1":285,"temp2":280,"temp3":false,"temp4":false,"state":254,"vehicle":1,"colour":6,"manual_override":0,"freeram":208540,"divertmode":1,"srssi":-51}
{"divert_update":0,"grid_ie":54}
{"divert_update":0,"grid_ie":51}
{"divert_update":0,"grid_ie":45}
{"divert_update":0,"grid_ie":38}
{"divert_update":0,"grid_ie":31}
{"divert_update":0,"grid_ie":34}
Time1
{"amp":0,"voltage":240,"pilot":8,"wh":0,"temp":285,"temp1":285,"temp2":281,"temp3":false,"temp4":false,"state":254,"vehicle":1,"colour":6,"manual_override":0,"freeram":213616,"divertmode":1,"srssi":-52}
{"divert_update":0,"grid_ie":36}
{"divert_update":0,"grid_ie":35}
{"divert_update":0,"grid_ie":34}
[... 324 lines cut ...]
$OK^20
$SV 245377^01
$OK^20
$FP 0 1 Time  3:00:07 PM^2F
$OK^20
$FP 0 1 Time  3:00:08 PM^20
$OK^20
$FP 0 1 Time  3:00:09 PM^21
$OK^20
$FP 0 1 Time  3:00:10 PM^29
$OK^20
$FP 0 1 Time  3:00:11 PM^28
$OK^20
$FP 0 1 Date 02/06/2021^02
$OK^20
$FP 15 1       ^27
$OK^20
$SV 245374^02
$OK^20
$FP 0 1 Time  3:00:15 PM^2C
$OK^20
$FP 0 1 Time  3:00:16 PM^2F
$OK^20
$FP 0 1 Time  3:00:17 PM^2E
$OK^20

js1tr3, 2021-06-02:
is there a good way to get the config?

js1tr3, 2021-06-02:
I am using solar divert over mqtt, but it didnt charge with divert in either state.

FatJuan, 2021-06-03:
This is the same issue as we are having GUI not available after Wifi restart #191 I am using V5 hardware and Huzzah ESP32 as well.

glynhudson, 2021-06-03:
>is there a good way to get the config?

The config is from `http://openevse-XXX/config` 

Are you sure you uploaded the correct FW for the Huzzah? This is `openevse_huzzah32.bin` https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/download/4.0.1/openevse_huzzah32.bin

`firmware.bin` is for the OpenEVSE WiFi V1 module.

[image]

treepleks, 2021-06-06:
To reassure people around, I just upgraded my Huzzah-ESP32 based EVSE wallbox with 7.1.3 on the ATMega and 4.0.1 on the Huzzah: everything seems to be working fine. I just tried to charge the car and it does the job. The Wifi3 web interface initially reported the protocol as 5.1.3 (or something similar) but after flashing, it now reports nothing (a dash).

js1tr3, 2021-06-06:
here is my config (back on V4, still wont charge, gets two about 2 amps then goes to sleep).

<li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li><li class="tabs-menu-item rawdata " role="presentation"><span class="devtools-tab-line"></span></li><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div class="toolbar"><div class="devtools-separator"></div><input class="searchBox devtools-filterinput" placeholder="Filter JSON"></div><div class="panelContent">
&nbsp; | &nbsp;
-- | --
firmware | "7.1.3"
protocol | "-"
espflash | 4194304
version | "4.0.1.dev"
diodet | 0
gfcit | 0
groundt | 0
relayt | 0
ventt | 0
tempt | 0
service | 1
scale | 0
offset | 0
mqtt_supported_protocols | &nbsp;
0 | "mqtt"
1 | "mqtts"
http_supported_protocols | &nbsp;
0 | "http"
1 | "https"
ssid | "intothesea"
pass | "_DUMMY_PASSWORD"
www_username | ""
www_password | ""
hostname | "openevse-2b8c"
sntp_hostname | "pool.ntp.org"
time_zone | "America/Detroit\|EST5EDT,M3.2.0,M11.1.0"
emoncms_server | "https://data.openevse.com/emoncms"
emoncms_node | ""
emoncms_apikey | ""
emoncms_fingerprint | ""
mqtt_server | "192.168.1.16"
mqtt_port | 1883
mqtt_topic | "emon/openevse"
mqtt_user | "emonpi"
mqtt_pass | "_DUMMY_PASSWORD"
mqtt_solar | ""
mqtt_grid_ie | "emon/Envoy/PowerNet"
mqtt_vrms | "emon/emonpi/vrms"
mqtt_announce_topic | "openevse/announce/2b8c"
ohm | ""
divert_PV_ratio | 1.1
divert_attack_smoothing_factor | 0.4
divert_decay_smoothing_factor | 0.05
divert_min_charge_time | 600
tesla_username | ""
tesla_password | ""
tesla_vehidx | -1
flags | 8714
emoncms_enabled | false
mqtt_enabled | true
mqtt_reject_unauthorized | true
ohm_enabled | false
sntp_enabled | true
tesla_enabled | false
divert_enabled | true
pause_uses_disabled | true
mqtt_protocol | "mqtt"
charge_mode | "fast"

</div>

js1tr3, 2021-06-06:
Here is the log from where it started charging. and then stops immediately.
$GG^24
$OK 0 228603^1D
$GU^36
$OK 1924 8987^20
$SV 228631^0D
$OK^20
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 1320 228631^2C
$FP 0 0 Charging 1.32A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 1320 228631^2C
$GU^36
$OK 2413 8987^2A
$FP 0 0 Charging 1.32A^52
$OK^20
$FP 14 0       ^27
$OK^20
$FP 0 1 Lifetime 9kWh^6D
$OK^20
$FP 13 1       ^21
$OK^20
$GG^24
$OK 1320 228631^2C
$FP 0 0 Charging 1.32A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 1320 228631^2C
$GU^36
$OK 2977 8987^25
$FP 0 0 Charging 1.32A^52
$OK^20
$FP 14 0       ^27
$OK^20
$SV 228804^05
$OK^20
$GG^24
$OK 1320 228804^24
$FP 0 0 Charging 1.32A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 0 228804^14
$GU^36
$OK 3382 8987^24
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$FP 0 1 EVSE Temp 44.2C^65
$OK^20
$FP 15 1       ^27
$OK^20
$GG^24
$OK 0 228804^14
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 0 228804^14
$GU^36
$OK 3382 8987^24
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 0 228804^14
$FP 0 0 Charging 0.00A^52

FatJuan, 2021-06-07:
I rolled back my firmware to 7.0.2.EU and wifi to V3.3.2.  I had the same problems as previously. No connection on then off, driving me mad!   I then changed the wireless AP from a Draytek mesh 912c to a Draytek AP900 (non mesh standard AP). Result, all issues cured.  I cannot explain this.  I am going to upgrade to latest and see if it keeps working OK I will report back.

glynhudson, 2021-06-07:
> version	"4.0.1.dev"

I notice your running `4.0.1.dev` the pre-compiled binaries are developer versions. This indicates that maybe you are compiling yourself? Try using the pre-compiled versions of 4.0.1 which can be downloaded from the assets section of the release page. The 4.0.1 binaries have been proven to work correctly, we have been shipping 4.0.1 on all new units for several months now without any issues: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/4.0.1

FatJuan, 2021-06-08:
I reloaded my firmware to 7.1.3.EU and wifi to V4.0.1.  No issues at all.  Could I suggest users try a different AP if they can. It was my Draytek mesh 912c that was seemingly the problem. Happy charging!

js1tr3, 2021-06-08:
I've tried both, the pre-compiled and built versions. same result. flashed back bare metal and trying again. will report back

js1tr3, 2021-06-09:
here is the config on V4.0.1 flashed from bare metal.
{"firmware":"7.1.3","protocol":"-","espflash":4194304,"version":"4.0.1","diodet":0,"gfcit":0,"groundt":0,"relayt":0,"ventt":0,"tempt":0,"service":2,"scale":220,"offset":0,"mqtt_supported_protocols":["mqtt","mqtts"],"http_supported_protocols":["http","https"],"ssid":"intothesea","pass":"_DUMMY_PASSWORD","www_username":"","www_password":"","hostname":"openevse-2b8c","sntp_hostname":"pool.ntp.org","time_zone":"America/New_York|EST5EDT,M3.2.0,M11.1.0","emoncms_server":"https://data.openevse.com/emoncms","emoncms_node":"","emoncms_apikey":"","emoncms_fingerprint":"","mqtt_server":"192.168.1.16","mqtt_port":1883,"mqtt_topic":"emon/openevse","mqtt_user":"emonpi","mqtt_pass":"_DUMMY_PASSWORD","mqtt_solar":"","mqtt_grid_ie":"emon/Envoy/PowerNet","mqtt_vrms":"emon/emonpi/vrms","mqtt_announce_topic":"openevse/announce/2b8c","ohm":"","divert_PV_ratio":1.1,"divert_attack_smoothing_factor":0.4,"divert_decay_smoothing_factor":0.05,"divert_min_charge_time":600,"tesla_username":"","tesla_password":"","tesla_vehidx":-1,"led_brightness":128,"flags":1546,"emoncms_enabled":false,"mqtt_enabled":true,"mqtt_reject_unauthorized":true,"ohm_enabled":false,"sntp_enabled":true,"tesla_enabled":false,"divert_enabled":true,"pause_uses_disabled":false,"mqtt_protocol":"mqtt","charge_mode":"eco"}

js1tr3, 2021-06-09:
even with timers cleared and divert off, it just waits. 
[image]

js1tr3, 2021-06-09:
Still the same problem on latest precompiled firmware
$SV 239349^07
$OK^20
$FE^27
$OK^20
$GS^30
$AT 02 02 14 0701^32
$OK 02 0 02 0701^16
$FP 0 1 Time  4:57:50 PM^28
$OK^20
$FB 3^33
$OK^20
$FP 0 0 Connected 14A^29
$OK^20
$FP 13 0       ^20
$OK^20
$FP 0 1 Energy 0.00Wh^30
$OK^20
$FP 13 1       ^21
$OK^20
$AT 03 03 14 0541^34
$FB 6^36
$OK^20
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$SV 238967^00
$OK^20
$GG^24
$OK 0 238967^11
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 0 238967^11
$GU^36
$OK 0 8987^1E
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$FP 0 1 Lifetime 9kWh^6D
$OK^20
$FP 13 1       ^21
$OK^20
$GG^24
$OK 0 238967^11
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 0 238967^11
$GU^36
$OK 0 8987^1E
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 0 238967^11
$SV 239097^07
$OK^20
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 0 239097^16
$GU^36
$OK 0 8987^1E
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$FP 0 1 EVSE Temp 36.7C^65
$OK^20
$FP 15 1       ^27
$OK^20
$GG^24
$OK 0 239097^16
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 0 239097^16
$GU^36
$OK 0 8987^1E
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GG^24
$OK 0 239097^16
$FP 0 0 Charging 0.00A^52
$OK^20
$FP 14 0       ^27
$OK^20
$GS^30
$OK fe 0 03 0501^14
$GG^24
$OK 0 239097^16
$GP^33
$OK 367 360 -2560^2B
$GU^36
$OK 0 8987^1E
$FB 6^36
$OK^20
$FP 0 0 zzZ Sleeping Zzz^3D
$OK^20
$FP 0 1 Time  4:58:01 PM^23
$OK^20
$SV 239235^0D
$OK^20
$FP 0 1 Time  4:58:02 PM^20
$OK^20
$FP 0 1 Time  4:58:03 PM^21
$OK^20
$FP 0 1 Time  4:58:04 PM^26
$OK^20
$FP 0 1 Time  4:58:05 PM^27
$OK^20
$FP 0 1 Date 09/06/2021^09
$OK^20
$FP 15 1       ^27
$OK^20
$FP 0 1 Time  4:58:09 PM^2B
$OK^20
$FP 0 1 Time  4:58:10 PM^23
$OK^20
$FP 0 1 Time  4:58:11 PM^22
$OK^20
$SV 239103^0B
$OK^20
$FP 0 1 Time  4:58:12 PM^21
$OK^20
$FP 0 1 Time  4:58:13 PM^20
$OK^20
$FP 0 1 Date 09/06/2021^09
$OK^20
$FP 15 1       ^27
$OK^20
$SV 239104^0C
$OK^20
$FP 0 1 Time  4:58:17 PM^24
$OK^20
$FP 0 1 Time  4:58:18 PM^2B
$OK^20

js1tr3, 2021-06-10:
If I uncheck solar divert it works!

js1tr3, 2021-06-10:
re-checked solar divert, clicked save and now it works. must have been some stuck setting..
[image]

js1tr3, 2021-06-11:
[image]
Its still acting wonky, EV connected flipping to state 254 and not charging. 
played with the divert settings swapping pages and it starts working but doesn't stay.

glynhudson, 2021-06-11:
So with solar PV divert switched off, the EVSE works and charges as expected? So this issue can be closed? Please open another issue if you are having issues with solar PV divert. 

Please also endure Emoncms demand shaper is switched off (V4.x is not supported yet by Demand Shaper) and try clearing any old residual timers with RAPI command `$ST 00 00 00 00`.

js1tr3, 2021-06-12:
Yup i did clear the timers, now I can't get it to stop charging with the pause button on the web interface... but at least I can unplug. And it follows the sun well. Even if it keeps going at 6A long after sundown. Thanks for the help debugging.  Now that I know how to compile I'll try to check out the code also.

closed 2021-06-12 by js1tr3