OpenEVSE/openevse_esp32_firmware #194: NTP not automatically working

glynhudson, 2021-06-04:
Many hours after a power outage I noticed the time on the EVSE was incorrect. NTP was enabled and setup correctly. The unit required me to click 'Set Time' to force an NTP update to correct the time: 

[image]

FW: V4.0.1

rvoosterhout, 2021-06-28:
I have had the same issue. NTP didn't update the time directly. If I use the browser method, it does change the time, but after 2 seconds it changes the time 2 hours in the past.

jeremypoulter, 2021-07-18:
This potentially could be related to #212, can you re-test after merging?

jeremypoulter, 2021-07-18:
No ignore that, been able to reproduce...