OpenEVSE/openevse_esp32_firmware #195: nodemcu-32s is panic looping

ronnyf, 2021-06-05:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:844
load:0x40078000,len:8140
ho 0 tail 12 room 4
load:0x40080400,len:5732
entry 0x4008069c

OpenEVSE WiFI 8d80
Firmware: 4.0.1.dev
Build date: Jun  4 2021 20:51:38
IDF version: 3.30300.190916
Free: 306968
No JSON config found, trying v1 settings
Loading config
Got '' 255 == 128 @ 575:32
Got '' 255 == 128 @ 0:32
Got '' 255 == 128 @ 32:64
Got '' 255 == 128 @ 542:33
Got '' 255 == 128 @ 128:45
Got '' 255 == 128 @ 173:32
Got '' 255 == 128 @ 438:60
Got '' 255 == 128 @ 732:96
Got '' 255 == 128 @ 205:45
Got '' 255 == 128 @ 250:32
Got '' 255 == 128 @ 282:32
Got '' 255 == 128 @ 314:64
Got '' 255 == 128 @ 378:30
Got '' 255 == 128 @ 408:30
Got 'ffffff'  255 == 127 @ 828:4
Got '' 255 == 128 @ 498:15
Got '' 255 == 128 @ 513:15
Got '' 255 == 128 @ 528:10
Got 'ffffff'  255 == 127 @ 538:4
Got '' 255 == 128 @ 607:45
Got '' 255 == 128 @ 652:80
After config_load_settings: 304532
[E][vfs_api.cpp:64] open(): /littlefs/schedule.json does not exist
Starting AP
(WIFI_MODE_AP == (WiFi.getMode() & WIFI_MODE_AP)) = 0
(WIFI_MODE_STA == (WiFi.getMode() & WIFI_MODE_STA)) = 0
Got Network event SYSTEM_EVENT_WIFI_READY
Got Network event SYSTEM_EVENT_AP_START
Set host name to openevse-8d80
Got Network event SYSTEM_EVENT_AP_START
Set host name to openevse-8d80
Got Network event SYSTEM_EVENT_AP_START
Set host name to openevse-8d80
AP IP Address: 192.168.4.1
After net_setup: 243128
Server started
After web_server_setup: 237556
Button pressed
{"state":1,"vehicle":0,"colour":2,"manual_override":0}
first read RAPI values
Checking for existing timers
{"amp":0,"voltage":240,"pilot":24,"wh":0,"temp":305,"temp1":305,"temp2":false,"temp3":false,"state":1,"vehicle":0,"colour":2,"manual_override":0,"freeram":232600,"divertmode":1,"srssi":0}
start_hour = 0
start_minute = 0
end_hour = 0
end_minute = 0
No timers on EVSE board
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400da2d1  PS      : 0x00060230  A0      : 0x800da8d4  A1      : 0x3ffb1e60  
A2      : 0x3ffc18f4  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x3ff47024  A7      : 0x00000000  A8      : 0x800da2c9  A9      : 0x3ffb1e40  
A10     : 0x00000bb1  A11     : 0x00000012  A12     : 0x00000000  A13     : 0x0000000f  
A14     : 0x40184590  A15     : 0x401845a4  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c428  LEND    : 0x4000c42d  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400da2d1:0x3ffb1e60 0x400da8d1:0x3ffb1e80 0x400f0135:0x3ffb1eb0 0x400da0a1:0x3ffb1ed0 0x400da0fb:0x3ffb1ef0 0x400da112:0x3ffb1f20 0x400dcfed:0x3ffb1f40 0x400dab27:0x3ffb1f70 0x4017fa60:0x3ffb1fb0 0x40088455:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:844
load:0x40078000,len:8140
ho 0 tail 12 room 4
```

uploaded via platformIO; config:
```[env:openevse_nodemcu-32s]
platform = ${common.platform}
board = nodemcu-32s
framework = arduino, espidf
lib_deps = ${common.lib_deps}
src_build_flags =
  ${common.version}.dev
  ${common.src_build_flags}
  ${common.debug_flags}
  -DWIFI_LED=LED_BUILTIN
  -DWIFI_LED_ON_STATE=LOW
  -DRAPI_PORT=Serial2
  -DI2C_SDA=21
  -DI2C_SCL=22
build_flags =
  ${common.build_flags}
  -DRX1=16
  -DTX1=17
#upload_port = openevse.local
upload_port = /dev/cu.usbserial-0001
upload_speed = 921600
monitor_speed = 115200
extra_scripts = ${common.extra_scripts}
board_build.partitions = ${common.build_partitions}
```
RX/TX (yellow, green) connected to GPIO16/17

ronnyf, 2021-06-06:
Reverting back to the 3.3 branch did work for now with some additional settings.

glynhudson, 2021-06-07:
Try doing a full erase before uploading `esptool.py erase_flash`

I notice your running `4.0.1.dev` the pre-compiled binaries are developer versions. This indicates that maybe you are compiling yourself? Try using the pre-compiled versions of 4.0.1 which can be downloaded from the assets section of the release page. The 4.0.1 binaries have been proven to work correctly, we have been shipping 4.0.1 on all new units for several months now without any issues: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/4.0.1

ronnyf, 2021-06-07:
I did both; the erase and even 4x write of the 1MB zero file ... neither worked

glynhudson, 2021-06-07:
Interesting, the debug output implies that an older config is still present in the EEPROM

ronnyf, 2021-06-07:
I will retry with another device and will post results here.

glynhudson, 2021-06-07:
Are you running OpenEVSE controller V7.1.2 or above FW https://github.com/openenergymonitor/open_evse/releases

ronnyf, 2021-06-07:
> Are you running OpenEVSE controller firmware V7.1.2 or above FW https://github.com/openenergymonitor/open_evse/releases. This is required for V4.x WiFi FW

Yes, I upgraded the controller just prior successfully

chaseadam, 2021-11-23:
I am getting what I believe is a similar error.

The `esptool.py erase_flash` suggestion did not remedy the situation.

```
...
                       AP IP Address: 192.168.4.1
After net_setup: 240128
                       Server started
After web_server_setup: 233736
                              {"amp":0,"voltage":240,"pilot":12,"wh":0,"session_energy":0,"total_energy":0,"temp":182,"temp_max":182,"temp1":182,"temp2":false,"temp3":false,"state":0,"flags":0,"vehicle":0,"colour":0,"manual_override":0,"freeram":228916,"divertmode":1,"srssi":0}
{"state":1,"flags":512,"vehicle":0,"colour":2}
first read RAPI values
Checking for existing timers
start_hour = 0
start_minute = 0
end_hour = 0
end_minute = 0
No timers on EVSE board
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dd5c9  PS      : 0x00060430  A0      : 0x800ddc50  A1      : 0x3ffb1e60
A2      : 0x3ffc1a08  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000000
A6      : 0x3ff47024  A7      : 0x00000000  A8      : 0x800dd5c1  A9      : 0x3ffb1e40
A10     : 0x000007d6  A11     : 0x00000012  A12     : 0x00000000  A13     : 0x0000000f
A14     : 0x40195b0c  A15     : 0x40195b20  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c428  LEND    : 0x4000c42d  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400dd5c9:0x3ffb1e60 0x400ddc4d:0x3ffb1e80 0x400f4cc5:0x3ffb1eb0 0x400dd361:0x3ffb1ed0 0x400dd3bb:0x3ffb1ef0 0x400dd3d2:0x3ffb1f20 0x400dffb1:0x3ffb1f40 0x400ddeef:0x3ffb1f70 0x4019104c:0x3ffb1fb0 0x4008c561:0x3ffb1fd0

Rebooting...
```

I did notice that it did *not* backtrace if I connected the serial port a few seconds after wifi module startup.

```
                       AP IP Address: 192.168.4.1
After net_setup: 240092
                       Server started
After web_server_setup: 233788
                              OpenEVSE not responding or not connected
{"amp":0,"voltage":240,"pilot":12,"wh":0,"session_energy":0,"total_energy":0,"temp":182,"temp_max":182,"temp1":182,"temp2":false,"temp3":false,"state":0,"flags":0,"vehicle":0,"colour":0,"manual_override":0,"freeram":229232,"divertmode":1,"srssi":0}
{"state":1,"flags":512,"vehicle":0,"colour":2}
first read RAPI values
Checking for existing timers
start_hour = 0
start_minute = 0
end_hour = 0
end_minute = 0
No timers on EVSE board
{"amp":0,"voltage":240,"pilot":12,"wh":0,"session_energy":0,"total_energy":0,"temp":182,"temp_max":182,"temp1":182,"temp2":false,"temp3":false,"state":1,"flags":512,"vehicle":0,"colour":2,"manual_override":0,"freeram":228088,"divertmode":1,"srssi":0}
{"amp":0,"voltage":240,"pilot":12,"wh":0,"session_energy":0,"total_energy":0,"temp":182,"temp_max":182,"temp1":182,"temp2":false,"temp3":false,"state":1,"flags":512,"vehicle":0,"colour":2,"manual_override":0,"freeram":227856,"divertmode":1,"srssi":0}
```

chaseadam, 2021-11-23:
I can also confirm that tag 3.3.2 works without issue.

jeremypoulter, 2021-11-27:
I was not able to replicate this, but may be related to #251 and #254 so should be resolved by #274 and #275

jeremypoulter, 2022-01-05:
Those PRs have been merged @ronnyf are you able to test with the latest version?

chaseadam, 2022-01-07:
I can confirm an unmodified checkout of `tag/4.1.1` works via web upload from version 3.3.2 without issue. Love that no settings reset required (is it recommended?). Thanks!

Note: shows as 4.1.0.dev in the version because version bump in code came after tag

ronnyf, 2022-01-07:
> Those PRs have been merged @ronnyf are you able to test with the latest version?

looking into it this weekend ...

ronnyf, 2022-01-08:
I'm able to get the nodemcu to run without panic looping on tag 4.1.1; I see there's a problem with the web interface, the current table is empty:

```

Service Level: | -1
-- | --
Level -1 Minimum: | -1 A
Level -1 Maximum: | -1 A
Sensor Scale: | 220
Sensor Offset: | 0
```

though I am seeing good log output from the openevse:
```
{"amp":0,"voltage":240,"pilot":32,"wh":2828979,"session_energy":0,"total_energy":2828.979,"temp":232,"temp_max":232,"temp1":232,"temp2":false,"temp3":false,"state":1,"flags":512,"vehicle":0,"colour":2,"manual_override":0,"freeram":227624,"divertmode":1,"srssi":-41}
```

ronnyf, 2022-01-08:
@jeremypoulter I reverted back to 3.2.2 which works fine ...

jeremypoulter, 2022-01-09:
@ronnyf I think something is failing to load if you can check the browser's developer tools to see if there are any errors?

ronnyf, 2022-01-09:
> 

something is definitely wrong ... browser fails to save config for mostly everything ...

jeremypoulter, 2022-01-09:
Which build is that with? I fixed that issue a while ago I thought...

jeremypoulter, 2022-01-09:
And what is the URL you are using to load the page?

ronnyf, 2022-01-09:
> Which build is that with? I fixed that issue a while ago I thought...

since the web interface had a bit of an issue on 4.1.1, I reverted back to the latest 3.3.2; 

this is what's happening when I try to setup MQTT:

Request
```
POST /config
Content-Type: application/json
Accept: */*
Origin: http://openevse-8d80.local
Referer: http://openevse-8d80.local/
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.2 Safari/605.1.15
```

Response
```
NaN null
```

When I run (copy as curl in web inspector) this as a curl, I am getting a good response:
```
curl 'http://192.168.68.77/config' \
-X 'POST' \
-H 'Content-Type: application/json' \
-H 'Accept: */*' \
-H 'Origin: http://openevse-8d80.local' \
-H 'Referer: http://openevse-8d80.local/' \
-H 'User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.2 Safari/605.1.15' \
--data-binary '{"mqtt_enabled":true .....
```

ronnyf, 2022-01-09:
> And what is the URL you are using to load the page?

http://openevse-8d80.local

ronnyf, 2022-01-09:
@jeremypoulter 
just to clarify that I am doing the right thing when deploying the source... I am 
1: checking out the 4.1.1 tag of the main repo; 
2: then master on the gui submodule 
3: and master on gui's posix_tz_db 

then I build this on platform io and upload (with the slight modification of the upload port)

jeremypoulter, 2022-01-09:
That should work although you can just do `git submodule update --init --recursive` from the tip level. Also unless you build the GUI, `npm run build` in the GUI dir, the pre-built GUI is used.

ronnyf, 2022-01-09:
@jeremypoulter So here's the deal; I am an idiot! My dev directory was mapped to iCloud Drive and this had some weird side effects on the build. I started over on a fresh, new (unmapped) directory and ... voila ... all good on 4.1.1 - nodemcu. Thanks for all the help.

jeremypoulter, 2022-01-10:
Cool, glad to hear