OpenEVSE/openevse_esp32_firmware #196: Log Solar PV divert MQTT data to Emoncms

glynhudson, 2021-06-07:
In order to help with debugging, it would be very useful if the OpenEVSE could log the MQTT data being received for Solar PV Gen / Grid IE to Emoncms along with the calculated current value. Solar PV divert operation seems to cause a lot of confusing, it's very hard to debug without data being recorded and graphed since screenshots of the interface don't always show the full picture since the data is filtered.

js1tr3, 2021-06-09:
like this?
[image]

jeremypoulter, 2021-06-09:
Thanks, I thought I had done it just hadn't got round to checking ;-) @glynhudson everything that updates in realtime on the UI is also exposed over MQTT

glynhudson, 2021-06-09:
Mm I'm running latest 4.0.1 and posting to emoncms.org with solar divert enabled and I don't see many of the Inputs you have. What version are you running? 

[image]

js1tr3, 2021-06-10:
I'm running 4.0.1, but I have my own emoncms server running at home with the latest version from git.

jeremypoulter, 2021-06-10:
Ah right it is only goes to MQTT, not EmonCMS, can probably make it do both, but is more event driven so didn't send to EmonCMS initially to avoid sending too many messages

glynhudson, 2021-06-17:
Ah right, that's the issue. If we could post all the metrics to emoncms that would be ace, most users use the Emoncms service to log the data to Emoncms rather than MQTT

closed 2021-07-15 by glynhudson