OpenEVSE/openevse_esp32_firmware #199: Eco mode and charge continues with no solar

lebsd, 2021-06-12:
Openevse V7.1.3 and Wifi V4.0.0., set in Eco mode and monitoring demand from grid (MQQT updated every 9 min).  Required PV power ratio:
0.2, smoothing attack 0.4, smoothing decay:
0.05, Minimum charge time: 600.  Connected Tesla Y in the afternoon, charge started at 8A, and as solar is reduced, it went to 6A. 

Issue:  It was already dark, around 9 PM and car was still charging at 6A when according to doc, for Eco mode, it should stop when solar <6A (Mqqt was 1800 W and not negative).

To stop the charge, I set the mode to normal and then manually pause.  Then switch mode to Eco again for next day, showing awaiting solar, as expected.  

It will be nice if either the start or stop time could be set when in Eco mode.  Or perhaps both a start and stop time. This will be helpful specially with days when in the AM will be semi cloudy, but then the afternoon is sunny, ex: setting a start to monitor solar at 1PM ending monitoring at sunset.  Or just able to set a Stop to avoid charging after certain time in a day.

jeremypoulter, 2021-06-12:
Updating every 9 minutes is really slow for divert, can you change the update rate of the MQTT feed? Ideally > 1 minute.

If that is not possible I would suggest setting the smoothing attack/delay to 1 as hopefully that is already done by your monitoring system and even if not at that update rate the smoothing will be useless.

treepleks, 2021-06-12:
Running 7.1.3 with 4.0.1 and I have the same behavior. I kept the original smoothing parameters (0.4 attack and 0.05 decay). The Eco mode starts charging correctly when Grid power import/export becomes sufficient but at the end of the day, the charge remains at 6A even when sun is going down.  My MQTT messages are frequent (every 10" roughly). The funny thing is that he messages in the main "OpenEVSE" Tab interface, in advanced mode, the session status window in the Wifi interface says that it is charging with decreasing current: 5A then 4A then 3A then 2A..., which should not be feasible. The Pilot remains at 6A. I will look into the code, this looks like a bad testing condition.

lebsd, 2021-06-12:
I have a rednode routine running on my NAS reading my meter.  I can adjust the sampling to any time.  The Eco recommends "5-10 sec" updates for "efficiency".  I can see ehy as the available current is adjusted more often.  However,  I am worried about the long term impact to the car battery, why I kept it at 9, which of course could mean that during the 9 min I may be pulling fm the grid, which I prefer as my car charge amps will be constant. I do not think that this update should affect when the charger should stop, it will just take 9 min to read that there is no solar vs 5 secs.  I am reviewing the divert.cpp code for the condition when it should stop.

jeremypoulter, 2021-06-12:
What is your concern about the effect on the battery?

The divert will switch off when the smoothed current is below 6A, with the default decay and a 9min update time, this will take a very long time, setting the attack/decay set to 1 it will effectively disable the smoothing so will turn off the charge as soon as the value you provide is less than 6A.

treepleks, 2021-06-12:
I just checked again, as the sun was slowly going down, and the charge didn't stop. After few hours spent below 5A (where it should already have stopped charging), the "charge rate" decreased slowly to 0 A (when the available current reached 0.99A, as expected) but the pilot still remained at 6, car charging. I reread the code of divert.cpp (I was the one that introduced the divert_PV_ratio in it), and compared it to Wifi3 version and found no striking difference in the logic (except for the "discussion" with the ATMega that uses a new protocol).   For some reason, the order to stop charging is not reaching the ATMega apparently. I used the web interface to turn Eco mode off and on again and it immediately stopped charging.

jeremypoulter, 2021-06-12:
Hum, that doesn't sound correct :( I think I need something to display the state that the WiFi controller thinks it should be in....

treepleks, 2021-06-12:
Jeremy is right, changing the charge rate dynamically should not damage the car battery nor the car charger which will be a solid state AC/DC converter. What damages Lithium batteries: 
- charging them to the maximum (100%), especially when it's hot 
- letting them completely discharge
- charging them with very high current (Very fast charging)
The best is to keep them around 60 to 75 % charge and to avoid fast charging too often, using DC chargers.

treepleks, 2021-06-12:
> Hum, that doesn't sound correct :( 
Indeed :-)

> I think I need something to display the state that the WiFi controller thinks it should be in....
If I can help, tell me. It's too late now, but tomorrow will be sunny again and I will be able to monitor again.

lebsd, 2021-06-12:
Jeremy, since I can control via my rednode the input as if it was the grid, I will simulate the sun going doing and monitor the evse charge and also the Mqtt values.  

I was looking at the divert.cpp code.  I do see now the smooth attack and smooth decay significance of the calculation for the charge rate shifting.  What I am unclear is what timing triggers divert.cpp, ie the mqtt update or the Min Charge Time at the Solar PV divert.  

Back to my original post of why the charge continues when there is no excess: based on the code, when the grid current is + (no excess), it should set the charge rate to 0 unless an mqtt_solar is not null, then charge rate=solar generation.  
  
      if (Igrid_ie < 0)
      {
        // If excess power
        double reserve = (1000.0 * ((divert_PV_ratio > 1.0) ? (divert_PV_ratio - 1.0) : 0.0)) / voltage;
        DBUGVAR(reserve);
        available_current = (-Igrid_ie - reserve);
      }
      else
      {
        // no excess, so use the min charge
        available_current = 0;
      }
    }
    else if (mqtt_solar!="")
    {
      // if grid feed is not available: charge rate = solar generation
      DBUGVAR(voltage);
      available_current = (double)solar / voltage;
    }

    if(available_current < 0) {
      available_current = 0;
    }
    DBUGVAR(available_current);

    double scale = (available_current > smoothed_available_current ? divert_attack_smoothing_factor : divert_decay_smoothing_factor);
    smoothed_available_current = (available_current * scale) + (smoothed_available_current * (1 - scale));
    DBUGVAR(smoothed_available_current);

    charge_rate = (int)floor(available_current);
    // if the remaining current can be used with a sufficient ratio of PV current in it, use it
    if ((available_current - charge_rate) > min(1.0, divert_PV_ratio)) {
      charge_rate += 1;

lebsd, 2021-06-12:
(Note:  I am not an expert CPP programmer, minimal experience).
Here are the results of the test (note the EV is connected but was not charging).  The EVSE shows "Waiting for solar" during all inputs. Seems based on divert.cpp calc are correct.  

Time    Grid(W)    Smooth     Available     Charge R      Max Current    Pilot Sensor
12:29    -1000     11.55A        4.17A             4A              Did not log      did not log
12:30    -100        11A            .42A              1A                  "                      6A
12:41      100        10.45A        0                   0                   "                       6A
12:43     -2800      10.94A        11.67A        12A                "                      12A
12:45     -2000       10.81A       8.33A          9A                "                         9A
12:47     -1200       10.52A      5A               5A                10A                    6A
12:49    945            9.99A         0                 0                  10A                   6A
 
Worth mentioning:  when I experienced the issue, the Evse state  was "not connected", then upon connecting, Evse state changed to "charging" even when <6A grid available and thereafter experienced that it will not stop at night.  So was looking at this part of the code
                        DBUGVAR(charge_rate);

    EvseState previousState = evse.getState();   <<<<<LUIS:  Is this correct: Evse previousState (ie a space between) ??

    // the smoothed current suffices to ensure a sufficient ratio of PV power
    if (smoothed_available_current >= (evse.getMinCurrent() * min(1.0, divert_PV_ratio)))
    {
      EvseProperties props(EvseState::Active);
      props.setChargeCurrent(charge_rate);
      evse.claim(EvseClient_OpenEVSE_Divert, EvseManager_Priority_Divert, props);

      if(previousState != evse.getState())
      {
        DBUGLN(F("Charge Started"));
        min_charge_end = divertmode_get_time() + divert_min_charge_time;
        event["divert_active"] = divert_active = true;
      }
    }

lebsd, 2021-06-21:
Quick update, set the MQTT update to 9 secs.  Now the unit does turn off correctly, in fact it now shuts off the charge way before the sun is completely out.  Thanks for the help and sorry for making you all look into it, although looking at the code helped me understand better the values of divert and decay, etc

closed 2021-06-21 by lebsd