OpenEVSE/openevse_esp32_firmware #203: Feature request: stop charging if no longer connected to MQTT broker or no updates received

jbakuwel, 2021-06-25:
We had an issue yesterday when plugging in the car resulted in 15A load on the domestic battery that went unnoticed for too long due to the fact that the software publishing available solar power stopped doing that. 

Something like a check box saying:

[ x ]  Disable charger after no update for 5 mins (5 mins is arbitrary: a deluxe version would make that configurable too)

Or something like:

Disable charger after no update for  ___ mins (0 minutes won't disable the charger)

I use other remote controlled devices via wifi that have a built-in fail safe (ie turn off loads) if they can no longer talk to the MQTT broker, which makes a lot of sense for off grid systems.

glynhudson, 2021-06-25:
I agree, that's a good idea

chris1howell, 2021-06-25:
I agree also, this would be a good enhancement.

pdhoogh, 2022-05-20:
I agree this is a good idea. In process control systems, such an option is called "shed mode". 

Consider the MQTT +I/-E to be an input to a controller, its output is the setpoint for the charge current control to the car. The +I/-E would be considered the Process Variable (PV) input to the controller. 

In professional control systems, when a PV no longer updates, it becomes BADPV represented by NaN (Not a Number). When the PV is NaN, the controller would freeze by default. Meaning that the charge current gets stuck where it is and no longer changes. The current functionality in the OpenEVSE is more or less like that?

To concur with industrial control systems standards, when the +I/-E no longer updates, the default action should be that the charge current setpoint is no longer updated, it stays where it is. 

Other shed modes may optionally be available. The above request to shut down the process is unusual in the case of industrial control. It would disrupt the industrial process too much. So it all depends on the viewpoint here. The above request would mean that when updates to +I/-E stop, the EV may not be ready for use and the way the owner would learn about this is with utter frustration finding that he/she will have to take the bike to wherever he/she is going. There may be a different answer.

One of the more typical shed mode options would be to revert to a "back-up control" that does not use +I/-E. For OpenEVSE that might be to shed the charge option to "Normal" with a preset current, e.g. 6 Amps. That would minimize the drain on the off-grid battery or the grid for that matter, while making sure the EV can still go somewhere.

Regardless of the chosen shed mode, industrial control systems help availability with automatic recovery. Meaning in this specific example that as soon as +I/-E is available again, normal control is automatically restored. So that a short glitch in the +I/-E updating does not mean your EV won't get charged until the owner figures that out by looking at a low EV battery level.

Food for thought...

closed 2026-09-21 by github-actions[bot]