OpenEVSE/openevse_esp32_firmware #211: Skoda Enyaq iV schedule charging issue

alenl2, 2021-07-14:
While scheduled charging is working great with my Kona the Enyaq does not like it. When you connect the cable the car locks the charging cable and waits for the power to come in. If no power comes within 15sec or so it will unlock the cable and go into "safe mode" charging. It will still charge when OpenEVSE closes the contactor on schedule but at a much slower rate (4kw). It also brakes Skoda's connected services for some reason.

Flipping the "Pause status" configuration doesn't fix it. So i left it on sleep.

The current workaround that works is to click the manual override button when in the first ~15sec of plugging in the car so the car charges for just 3sec then you can cancel the manual override. This fixes the "safe mode" charging and connected services.

I would suggest adding a configuration option so users can enable this compatibility mode like the "Pause status" configuration

glynhudson, 2021-07-22:
Oh dear, another car which doesn't follow the J1772 charging protocol. 

Both the Zoe and the MG ZS has this issue when they were first launched, on both vehicles this was subsequently fixed by the manufacturer. Maybe worth alerting Skoda of this bug, they may already have a fix? Not supporting timed charging and not waking up the on-board charger when commanded by the pilot signal is not great. I've not heard of any reports of this, but maybe the same issue will affect the ID3/4 since they share the same MEB platform?  

Just to confirm, a temporary work-around is to allow the car to charge for a few seconds after plugging in before pausing the charge, then the car will wake up and charge correctly many hours later when the timer schedule kicks in? Potentially, we could add this as a feature in the OpenEVSE, it could also be a good opportunity to measure how much current the EV draws in those few seconds which could be used to inform a smart charging schedule.

alenl2, 2021-07-22:
Yes exactly. If you just allow it to charge for around 5 seconds after plugging it in it charges correctly many hours later. 

If you don't the car enters an errored charging state (the same happens if the car can't engage the locking pin on the connector) and it will charge at a much slower rate (blinking red light). But if the car can charge for even a few seconds the charging light turns green and remains green even if you remove power.

Even the thermal management is ok with this workaround as the battery coolant pumps and fans will stop when the power is removed (awaiting schedule)

Come to think of it when i configure a scheduled charge on my Hyundai Kona (on the car itself) it will tell the charger also to engage the relay and charge the car for a few seconds when plugging it in and then wait for the proper time.

Maybe the car wants to be sure that the fuses/relays wont instantly explode 🙂

closed 2022-05-25 by glynhudson