OpenEVSE/openevse_esp32_firmware #218: Station Serial Number

chris1howell, 2021-07-19:
Devoloper Mode should allow a set once (If null) feature to assign the station and charge port serial number. Wiping would require full FLASH/EEPROM erase.

This serial number should be avaliable in the HTTP configuration info, MQTT announce and sent via OCPP.

jeremypoulter, 2021-07-20:
Is this something that should go into the OpenEVSE flash?

jeremypoulter, 2021-07-24:
@chris1howell Can you comment on if these should be in the OpenEVSE flash? Maybe just the meter serial number? but can you also clarify what the difference between each of these is?

Realistically how secure do these values need to be? Is just writing to the flash and blocking in software sufficient? or do we been to depend on hardware features to protect from changes, eg use the eFUSE on the ESP32?

glynhudson, 2021-07-26:
Can the ESP serial number be used for this? I guess this could cause issues if the WiFi module was changed / upgraded?

chris1howell, 2021-07-26:
The connector serial number should be the controller, same with the power
meter unless another external power meter is used. The chassis serial
number can be stored wherever we choose.

jeremypoulter, 2021-07-27:
I think in our context the meter is the OpenEVSE and also in the future we should have multiple OpenEVSE per WiFi module, so I would propose the following:

- ESP32 unique ID for the controller/station serial, can't be set
- Unique ID from the ATMEGA382 (https://www.thethingsnetwork.org/forum/t/arduino-has-a-unique-id/21415/4) for the meter id
- one user programmable id stored in (ESP32) flash for the port? Although again with multiple EVSEs would there logically be multiple? @chris1howell you also mentioned chassis number is this the same?

Is this acceptable?

jeremypoulter, 2021-07-27:
https://gist.github.com/speters/f889faec42b510052a6ab4be437d38ca

jeremypoulter, 2021-07-27:
https://github.com/ricaun/ArduinoUniqueID

PR #297 (2022-01-21): Retrieve the serial numbers of the WiFi module and EVSE module
Fixes #218 Station Serial Number
closed 2022-01-21 by glynhudson