OpenEVSE/openevse_esp32_firmware #219: OCPP - Send Station details

chris1howell, 2021-07-19:
Send Charge Station / Port Serial numbers, Firmware version and firmware timestamp, meter type and Serial Number via OCPP.

matth-x, 2021-08-03:
OCPP standardizes a handful of possible attributes in the provisioning UC. The best is to transmit as many of them as possible. The following table contains the excerpt of the specification with the relevant fields:

| Attribute | Description | Candidate? |
| -------------- | -------------- | :------: |
| chargePointModel | Required. This contains a value that identifies the model of the ChargePoint. | hardcoded `"Advanced Series"` |
| chargePointSerialNumber | Optional. This contains a value that identifies the serial number of the Charge Point. | see Issue #218 |
| chargePointVendor | Required. This contains a value that identifies the vendor of the Charge Point. | hardcoded `"OpenEVSE"` |
| firmwareVersion | Optional. This contains the firmware version of the Charge Point. | In `evse_man.h`: `getFirmwareVersion()` |
| meterSerialNumber | Optional. This contains the serial number of the main electrical meter of the Charge Point. | see Issue #218 |
| meterType | Optional. This contains the type of the main electrical meter of the Charge Point. | omit or hardcoded |

I've already extended the OCPP API to support all fields. Now, they only have to be filled with the respective data. Can you approve of the proposed candidates?

matth-x, 2022-01-09:
@jeremypoulter Thank you for your efforts in #297.

Which attributes do you finally have in mind for the chargePointSerialNumber and meterSerialNumber fields? What about

chargePointSerialNumber: `serial` (i.e. `ESPAL.getLongId()`)
meterSerialNumber: Maybe this field is only necessary for legal issues in combination with an external MID meter anyway. However we can use it for the Atmega board serial.

Furthermore, what about using `BUILD_TAG` as the firmwareVersion? I think for the OCPP backend that's the more relevant information.

Changing these fields in future won't be a hassle though.

jeremypoulter, 2022-01-09:
@matth-x I think `serial` for the charge point and the EVSE serial for the meter. The `currantfirmware` should be used for the firmware version, but that does resolve to `BUILD_TAG`

matth-x, 2022-01-10:
@jeremypoulter Thank you, then every field is clear. I will integrate and test this soon.

matth-x, 2022-02-01:
Integrated in matth-x/ESP32_WiFi_V4.x@110c05feb517df12b859f90aa88363cd3b27fc82. Works great!

closed 2022-02-01 by matth-x